mne-tools/mne-python #9374: Repeated Epoch cropping is not idempotent when include_tmax=False

jsosulski, 2021-05-06:
#### Describe the bug

Repeated cropping of `mne.Epoch` when `include_tmax=False` cuts off the last time sample.

#### Steps to reproduce

```python
import mne
import numpy as np

n_epochs, n_channels, n_timepoints = 10, 4, 21
ch_names = [f"ch_{i}" for i in range(n_channels)]
fake_data = np.random.random((n_epochs, n_channels, n_timepoints))

info = mne.create_info(ch_names=ch_names, sfreq=40)
epo = mne.EpochsArray(fake_data, info)

for i in range(10):
    epo.crop(tmin=0, tmax=0.4, include_tmax=False)
    print(f"{epo.tmax=}")
```

#### Expected results

The `epo.tmax` should be set to 0.375 on the first and all subsequent iterations.

#### Actual results

Every loop iteration cuts off the last time sample of the epoch.

#### Additional information

```
Platform:      Linux-5.4.0-72-generic-x86_64-with-glibc2.29
Python:        3.8.5 (default, Jan 27 2021, 15:41:15)  [GCC 9.3.0]
Executable:    /home/jan/dev/python/spot/spot_venv/bin/python
CPU:           x86_64: 8 cores
Memory:        Unavailable (requires "psutil" package)
mne:           0.24.dev0
numpy:         1.19.5 {blas=openblas, lapack=openblas}
scipy:         1.5.4
matplotlib:    3.3.4 {backend=module://backend_interagg}

sklearn:       0.23.2
numba:         Not found
nibabel:       3.2.1
nilearn:       Not found
dipy:          Not found
cupy:          Not found
pandas:        1.1.5
mayavi:        Not found
pyvista:       Not found
vtk:           Not found
```

The cause of the issue is that when the `tmax` argument for `crop` is out of range of the `Epoch`, it is set to `epo.tmax`. However, this behaviour is (at least for me) unintuitive when `include_tmax=False` as I do not want to cut off the last time sample, if my desired interval is larger than the epoch.

Possible solution would be to set `tmax` to `tmax + 1/sfreq` when `include_tmax=True` and the desired tmax-crop is outside the epoch interval.

agramfort, 2021-05-06:
the pb is here:

https://github.com/mne-tools/mne-python/blob/main/mne/utils/numerics.py#L475

we need remove the last time point even if we shouldn't

would you be willing to open a PR to fix this?

>

jsosulski, 2021-05-06:
Sure, I will submit a PR.

jsosulski, 2021-05-06:
I think (for now) the fix needs to be done in the outer call in the `crop` function of the Epoch class, as in the `_time_mask` function we do not have the information whether `tmax` was set automatically or not.

However, using `grep` I found several calls to `_time_mask` that could have similar issues as Epoch. Could check the relevant sections tomorrow though.

jsosulski, 2021-05-07:
For posterity: the same issue is present for `Evoked`

PR #9378 (2021-05-10): FIX: make epoch cropping idempotent
#### Reference issue
Fixes #9374 

#### What does this implement/fix?
Set `include_tmax=True` when `tmax` is larger than the epoch.

#### Additional Information
I think this cannot be sensibly done in the `_time_mask` function as we do not know whether `tmax` was set automatically or not in that function. However, I do now knot if `_time_mask` should do this kind of error handling or if it is up to the calling functions.
closed 2021-05-10 by larsoner