mne-tools/mne-python #9377: choice of thread-based vs process-based parallelism in time-freq representation

neerajwagh, 2021-05-06:
#### Describe the new feature or enhancement
Enhancement - Let the user control the type of parallelism (threading or processes) used under the hood by joblib, just like n_jobs currently is. It would be nice to have the user control whether mne creates processes or threads in the background, to make memory management easier/more predictable for user-written mne-based processing pipelines.

Although mne's parallel module at https://github.com/mne-tools/mne-python/blob/ce6e61c29462eee3ec7a0a9209e8d71f64e34745/mne/parallel.py#L103 accepts this choice, it is not explicitly passed to it in https://github.com/mne-tools/mne-python/blob/ce6e61c29462eee3ec7a0a9209e8d71f64e34745/mne/time_frequency/tfr.py#L390. 

**(As far as I understand, the user cannot pass the 'prefer' argument through an environment variable. Please let me know if I can do that, which makes this a non-issue and can be closed.)**

Passing prefer=None invokes joblib's default behavior of using processes.

I'm not sure if this behavior is specific to TFR, or it happens with other functions too.

#### Describe your proposed implementation
Update user-facing function definitions to include joblib's 'prefer' argument. Pass that through to mne's parallel module.

#### Describe possible alternatives
Using a global environment variable that a user can set if function definitions will be too cumbersome to change throughout the code base for all the mne functions using n_jobs as an input.

#### Additional comments
How this change would affect my personal use case - In my pipeline, I'm currently using concurrent.futures to process each recording in a separate process through a process pool. However, if n_jobs > 1, mne creates more processes than can fit in memory, and the pipeline fails. If I could choose to create joblib threads instead, I believe I can still remain within my memory limits since threads would share the memory with the parent process. (Let me know if I'm wrong). With n_jobs >1, the processes spawned by the pipeline+mne reaches the memory limit and crashes with BrokenProcessPool error.

agramfort, 2021-05-06:
have you considered a the joblib context manager eg

with parallel_backend('threading', n_jobs=2):

see example in https://joblib.readthedocs.io/en/latest/parallel.html

neerajwagh, 2021-05-07:
Oh I didn't know that could work. I'll try over the weekend, and share the results. You may close this if it is a non-issue. Thanks!

closed 2021-05-11 by larsoner