mne-tools/mne-python #9380: mne.io.read_raw_curry() cannot read the event file which file extension '.cdt.ceo'

dddd1007, 2021-05-07:
#### Describe the bug
I have tried to use `mne.io.read_raw_curry()` to read my data created by Curry 8, but there were no annotations in the Raw object. But EEGLAB can import the data and show the annotations correctly.

#### Steps to reproduce
I read the source code of [curry.py](https://github.com/mne-tools/mne-python/blob/maint/0.23/mne/io/curry/curry.py), and I found this function cannot handle the ".cdt.ceo" file. 

So I change the file extension of my file from ".cdt.ceo" to "cdt.cef", and the function works smoothly.

#### Expected results
`mne.io.read_raw_curry()` can read cef file too, or it will make some users confused.

#### Additional information
I am sorry for my weak program skill. I have tried to fix this bug.

drammock, 2021-05-07:
Can you share the original file (Dropbox, Google drive link, etc) so we can test?

dddd1007, 2021-05-07:
> Can you share the original file (Dropbox, Google drive link, etc) so we can test?

Definitely yes. Here is the original file.

https://www.dropbox.com/s/d41yfv7ik0ldzpv/eeg_raw.zip?dl=0

dddd1007, 2021-05-07:
This bug is fixed by @drammock .

drammock, 2021-05-07:
FYI, normally we leave issues open until the fixing pull request has been merged, in case during review the solution is found to be incorrect or incomplete.

dddd1007, 2021-05-07:
I am sorry for my brusque being.

drammock, 2021-05-07:
> I am sorry for my brusque being.

no problem!  Different software projects handle issues/PRs differently.  Thanks for reporting this bug and helping to fix it.

PR #9381 (2021-05-10): Reading events info from ceo file
#### Reference issue
Fixes #9380

#### What does this implement/fix?
The read_raw_curry() cannot read the event file from .ceo file. So I try to make the function recognize .ceo files, a type of event file generated by Curry.
PR #9712 (2021-09-08): FIX: Read events from ceo file.
#### Reference issue
Fixes #9380 
Improves #9381 

#### What does this implement/fix?
With #9381, 'ceo' files can be imported after renaming to 'cef'. This PR adds '.ceo' extension to event file extensions.
closed 2021-05-10 by larsoner