mne-tools/mne-python #9393: Ipyvtk_simple is not ipyvtk_link in 3D notebook viewer

adam2392, 2021-05-12:
#### Describe the bug
I am attempting to use the 3D backend notebook viewer for plotting an ecog brain. I get the following error:

```
---------------------------------------------------------------------------
ModuleNotFoundError                       Traceback (most recent call last)
~/Documents/sickkids/.venv/lib/python3.8/site-packages/pyvista/plotting/plotting.py in show(self, title, window_size, interactive, auto_close, interactive_update, full_screen, screenshot, return_img, cpos, use_ipyvtk, **kwargs)
   4323             try:
-> 4324                 from ipyvtk_simple.viewer import ViewInteractiveWidget
   4325             except ImportError:

ModuleNotFoundError: No module named 'ipyvtk_simple'

During handling of the above exception, another exception occurred:

ImportError                               Traceback (most recent call last)
<ipython-input-20-ea5cd7c5fb7d> in <module>
----> 1 fig = plot_alignment(
      2     raw.info,
      3     subject="E1",
      4     subjects_dir=subjects_dir,
      5     surfaces=["pial"],

<decorator-gen-151> in plot_alignment(info, trans, subject, subjects_dir, surfaces, coord_frame, meg, eeg, fwd, dig, ecog, src, mri_fiducials, bem, seeg, fnirs, show_axes, dbs, fig, interaction, verbose)

~/Documents/sickkids/.venv/lib/python3.8/site-packages/mne/viz/_3d.py in plot_alignment(***failed resolving arguments***)
   1098     renderer.set_camera(azimuth=90, elevation=90,
   1099                         distance=0.6, focalpoint=(0., 0., 0.))
-> 1100     renderer.show()
   1101     return renderer.scene()
   1102 

~/Documents/sickkids/.venv/lib/python3.8/site-packages/mne/viz/backends/_notebook.py in show(self)
    360         # viewer
    361         if LooseVersion(pyvista.__version__) < LooseVersion('0.30'):
--> 362             viewer = self.plotter.show(
    363                 use_ipyvtk=True, return_viewer=True)
    364         else:  # pyvista>=0.30.0

~/Documents/sickkids/.venv/lib/python3.8/site-packages/pyvista/plotting/plotting.py in show(self, title, window_size, interactive, auto_close, interactive_update, full_screen, screenshot, return_img, cpos, use_ipyvtk, **kwargs)
   4324                 from ipyvtk_simple.viewer import ViewInteractiveWidget
   4325             except ImportError:
-> 4326                 raise ImportError('Please install `ipyvtk_simple` to use this feature:'
   4327                                   '\thttps://github.com/Kitware/ipyvtk-simple')
   4328             # Have to leave the Plotter open for the widget to use

ImportError: Please install `ipyvtk_simple` to use this feature:	https://github.com/Kitware/ipyvtk-simple
```

However, this is because `ipyvtk_simple` is now named... `ipyvtk-link` :(

#### Steps to reproduce
Set `mne.viz.set_3d_backend("notebook")` and then run `plot_ecog.py` example.

#### Additional information
```
Platform:      macOS-10.16-x86_64-i386-64bit
Python:        3.8.6 | packaged by conda-forge | (default, Jan 25 2021, 23:22:12)  [Clang 11.0.1 ]
Executable:    /Users/adam2392/Documents/sickkids/.venv/bin/python
CPU:           i386: 8 cores
Memory:        Unavailable (requires "psutil" package)
mne:           0.24.dev0
numpy:         1.20.1 {blas=openblas, lapack=openblas}
scipy:         1.6.0
matplotlib:    3.3.4 {backend=agg}

sklearn:       0.24.1
numba:         0.52.0
nibabel:       3.2.1
nilearn:       Not found
dipy:          Not found
cupy:          Not found
pandas:        1.2.2
mayavi:        Not found
pyvista:       0.28.1 {pyvistaqt=0.4.0, OpenGL 4.1 ATI-4.4.17 via AMD Radeon Pro 560 OpenGL Engine}
vtk:           9.0.1
PyQt5:         5.15.2
```

Also ran w/ `mne==0.23`

larsoner, 2021-05-12:
This upgrade is a bit of a mess, we probably need to backport the update to maint/0.23. If you `pip install --upgrade pyvista` do things work with 0.24.dev0 / `main`?

adam2392, 2021-05-12:
I don't think so but will try it tomorrow and lyk!

larsoner, 2021-05-12:
[0.30.1 is the latest](https://pypi.org/project/pyvista/0.30.1/) and they are designed to be upgraded simultaneously so I'm at least a little bit optimistic :)

GuillaumeFavelier, 2021-05-13:
Ayaye... I think the best way to solve this is to udpate `pyvista` :sweat_smile:

adam2392, 2021-05-13:
Running it again:

```
Platform:      Linux-5.5.4-050504-generic-x86_64-with-glibc2.10
Python:        3.8.3 (default, May 19 2020, 18:47:26)  [GCC 7.3.0]
Executable:    /home/adam2392/Documents/sickkids/.venv/bin/python3
CPU:           x86_64: 24 cores
Memory:        Unavailable (requires "psutil" package)
mne:           0.23.0
numpy:         1.20.2 {blas=openblas, lapack=openblas}
scipy:         1.6.2
matplotlib:    3.4.1 {backend=module://ipympl.backend_nbagg}

sklearn:       0.24.1
numba:         0.53.1
nibabel:       3.2.1
nilearn:       Not found
dipy:          Not found
cupy:          Not found
pandas:        1.2.4
mayavi:        Not found
pyvista:       0.30.1 {OpenGL 4.5.0 NVIDIA 460.73.01 via GeForce GTX 750/PCIe/SSE2}
vtk:           9.0.1
PyQt5:         5.15.4
None
```

I get:

```
---------------------------------------------------------------------------
AttributeError                            Traceback (most recent call last)
<ipython-input-3-39066c44604d> in <module>
      1 get_ipython().run_line_magic('matplotlib', 'widget')
----> 2 fig = plot_alignment(raw.info, subject='fsaverage', subjects_dir=subjects_dir,
      3                      surfaces=['pial'], coord_frame='mri')
      4 az, el, focalpoint = 160, -70, [0.067, -0.040, 0.018]
      5 mne.viz.set_3d_view(fig, 

<decorator-gen-152> in plot_alignment(info, trans, subject, subjects_dir, surfaces, coord_frame, meg, eeg, fwd, dig, ecog, src, mri_fiducials, bem, seeg, fnirs, show_axes, dbs, fig, interaction, verbose)

~/Documents/sickkids/.venv/lib/python3.8/site-packages/mne/viz/_3d.py in plot_alignment(***failed resolving arguments***)
   1097     renderer.set_camera(azimuth=90, elevation=90,
   1098                         distance=0.6, focalpoint=(0., 0., 0.))
-> 1099     renderer.show()
   1100     return renderer.scene()
   1101 

~/Documents/sickkids/.venv/lib/python3.8/site-packages/mne/viz/backends/_notebook.py in show(self)
    350             viewer = self.plotter.show(
    351                 jupyter_backend="ipyvtk_simple", return_viewer=True)
--> 352         viewer.layout.width = None  # unlock the fixed layout
    353         # main widget
    354         if self._dock is None:

~/Documents/sickkids/.venv/lib/python3.8/site-packages/PIL/Image.py in __getattr__(self, name)
    539             )
    540             return self._category
--> 541         raise AttributeError(name)
    542 
    543     @property

AttributeError: layout
```

This is now on a linux machine too.

adam2392, 2021-05-13:
Oh just kidding, I guess maybe my ipyvklink isn't installing properly...

from running `pip install ipytklink`

```
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
    Preparing wheel metadata ... error
    ERROR: Command errored out with exit status 1:
     command: /home/adam2392/Documents/sickkids/.venv/bin/python3.8 /home/adam2392/Documents/sickkids/.venv/lib/python3.8/site-packages/pip/_vendor/pep517/_in_process.py prepare_metadata_for_build_wheel /tmp/tmpicomhibk
         cwd: /tmp/pip-install-zepgf25i/orjson
    Complete output (11 lines):
    💥 maturin failed
      Caused by: Cargo metadata failed. Does your crate compile with `cargo build`?
      Caused by: `cargo metadata` exited with an error: error: failed to run `rustc` to learn about target-specific information
    
    Caused by:
      process didn't exit successfully: `rustc - --crate-name ___ --print=file-names -Z mutable-noalias=yes -C target-feature=+sse2 --crate-type bin --crate-type rlib --crate-type dylib --crate-type cdylib --crate-type staticlib --crate-type proc-macro --print=sysroot --print=cfg` (exit code: 1)
      --- stderr
      error: the option `Z` is only accepted on the nightly compiler
    Error running maturin: Command '['maturin', 'pep517', 'write-dist-info', '--metadata-directory', '/tmp/pip-install-zepgf25i/orjson/pip-wheel-metadata', '--interpreter', '/home/adam2392/Documents/sickkids/.venv/bin/python3.8', '--manylinux=off', '--strip=on']' returned non-zero exit status 1.
    Checking for Rust toolchain....
    Running `maturin pep517 write-dist-info --metadata-directory /tmp/pip-install-zepgf25i/orjson/pip-wheel-metadata --interpreter /home/adam2392/Documents/sickkids/.venv/bin/python3.8 --manylinux=off --strip=on`
    ----------------------------------------
ERROR: Command errored out with exit status 1: /home/adam2392/Documents/sickkids/.venv/bin/python3.8 /home/adam2392/Documents/sickkids/.venv/lib/python3.8/site-packages/pip/_vendor/pep517/_in_process.py prepare_metadata_for_build_wheel /tmp/tmpicomhibk Check the logs for full command output.
```

adam2392, 2021-05-13:
Alternatively I tried on my mac:

I get: ```Error displaying widget: model not found``` 

Not sure if this is related to the original issue tho...?

[Untitled-2.pdf](https://github.com/mne-tools/mne-python/files/6474489/Untitled-2.pdf)

GuillaumeFavelier, 2021-05-17:
This is strange, I'm on linux myself and I never got this :thinking: 
Can you open an issue on https://github.com/Kitware/ipyvtklink?

GuillaumeFavelier, 2021-05-17:
By scrolling through the closed issue there I found https://github.com/Kitware/ipyvtklink/issues/27 which seems to have the same "model not found" issue, maybe it is related?

adam2392, 2021-05-21:
Done!

adam2392, 2021-06-09:
It seems there is no movement in the other repo regarding this. I think it's still having the same issue. I was wondering is there any way then to replicate something like `plot_ecog` on a Mac using Jupyter or such? Or any tips/suggestions?

Is there perhaps even an older version of pyvista/mne/etc. combo that might work?