mne-tools/mne-python #9400: How should data files with more than one measurement return data?

rob-luke, 2021-05-18:
Several fNIRS data file types can store than one measurement. For example SNIRF and Artinis files can store measurements from several participants.

Currently we do not support reading SNIRF files with more than one measurement. But I have received a request to allow reading of files with multiple measurements.

What is the best way to return the data? As a list of raws? And does any existing reader already do this or some flavour of this behaviour?

agramfort, 2021-05-18:
Yes I would return a list of raws

like the API of read_evokeds

There is no example in MNE for raw though

rob-luke, 2021-05-18:
Perfect, thanks @agramfort thats an ideal example for us to work from.

closed 2021-05-18 by rob-luke