mne-tools/mne-python #9409: Importing .SET Files returns Errors within Library Line 24

Thomasvandang, 2021-05-19:
Hi I'm trying to import my set file for analysis of erp and p300, but when reading the files, it returns these errors. 

Does anyone know what I'm doing wrong?

Code:

import mne
data_path = r"C:/Users/thoma/Desktop/EEG/session/1";
fname = data_path + '/EEG1.set'
mydata = mne.io.read_raw_eeglab(fname)

Error:

[image]

agramfort, 2021-05-19:
can you read the file with another  reader ?

https://pypi.org/project/pyEDFlib/

maybe in matlab? if not it's a like a pb of the file.

if mne is the pb please the file with me.

>

hoechenberger, 2021-05-25:
friendly ping @Thomasvandang, does the problem persist?

sappelhoff, 2021-08-08:
Closing, because there is too little information/data to know whether this is an MNE-Python problem. Please feel free to reopen this once you are available to share more information and the broken data file @Thomasvandang

closed 2021-08-08 by sappelhoff