mne-tools/mne-python #9415: ENH: Quantify location in label

larsoner, 2021-05-20:
I want to get a measure of how confident I can be that a given vertex within a label actually belongs to that label versus a neighboring one. One option is, for each vertex in the label, compute the minimum distance (in m) required to travel along the cortical surface before you reach a bounding vertex. I think this can be computed pretty easily using `dijkstra` in SciPy -- we just compute the minimum distance from each vertex in the label to the set of all edge vertices in the label. So maybe:
```
class Label:

    def distances_to_edge(self, subject=None, subjects_dir=None, surface='white'):
        """Compute the distance from each vertex to the edge of the label.
        ...
```
Then it might also be useful to have also a:
```
    def compute_area(self, subject=None, subjects_dir=None, surface='white')
```
or similar that computes the area of a given label. Then the distance can be compared to the total area (probably via a an equivalent-circle-radius `r = np.sqrt(a/np.pi)`) to give some sense of how close to the middle it likely is given the label's size, if you're more interested in relative rather than absolute distances.

agramfort, 2021-05-20:
I think what you want to do is to compute the Hausdorff distance between a
vertex and a label.

if it's constrained to the cortical surface you can compute with disjkstra
the distance between
your vertex and all the vertices in the surfaces with scipy.sparse taking
as input the adjacency
matrix of the vertices

>

larsoner, 2021-05-20:
I'm not sure how Hausdorff will help us here, since it has to do with maximal distances rather than minimal ones. Can you elaborate? I think you really want to know the minimal distance from a given vertex to the edge, not the maximal one.

But what you say about Dijkstra distance is what I had in mind in the above text, too.

PR #9424 (2021-07-15): MRG, ENH: Add label geometry methods
Closes #9415
closed 2021-07-15 by larsoner