mne-tools/mne-python #9417: Using multiple colors when multiple ICA components in the reconstructed latent source figure

crsegerie, 2021-05-21:
[image]

In the above figure, the colors look the same for me, and I don't know which component is which one.

marsipu, 2021-06-02:
Does this come from something like:

```python
ecg_evoked = create_ecg_epochs(raw).average()
ica.plot_sources(ecg_evokeds)
```
And it probably should only apply for evokeds?
I could look into it in the next time if nobody else does it yet

hoechenberger, 2021-06-02:
> Does this come from something like:
> 
> ```python
> ecg_evoked = create_ecg_epochs(raw).average()
> ica.plot_sources(ecg_evokeds)
> ```

Yep!

> And it probably should only apply for evokeds?

I suppose so, yes!

marsipu, 2021-06-03:
Would you want different colors just for the excluded components or for every component?

hoechenberger, 2021-06-03:
Only for the excludes, they are already labeled and appear in the legend

drammock, 2021-06-11:
closed by #9444

closed 2021-06-11 by drammock