mne-tools/mne-python #9418: Finding which channels are removed with compute_covariance

ilonadem, 2021-05-21:
I've been closely following the source localization tutorial from the MNE documentation (https://mne.tools/0.16/auto_tutorials/plot_mne_dspm_source_localization.html), but I have two questions about the `mne.compute_covariance` method. 

At the moment, I have that `raw = mne.io.read_raw_fif(raw_fname)` has 376 channels, 319 epochs, and 6 event IDs. When I call `print(raw.info)` I get: 
```
<Info | 15 non-empty values
 bads: 2 items (MEG 2443, EEG 053)
 ch_names: MEG 0113, MEG 0112, MEG 0111, MEG 0122, MEG 0123, MEG 0121, MEG ...
 chs: 204 GRAD, 102 MAG, 9 STIM, 60 EEG, 1 EOG
 custom_ref_applied: False
 dev_head_t: MEG device -> head transform
 dig: 146 items (3 Cardinal, 4 HPI, 61 EEG, 78 Extra)
 file_id: 4 items (dict)
 highpass: 0.1 Hz
 hpi_meas: 1 item (list)
 hpi_results: 1 item (list)
 lowpass: 40.0 Hz
 meas_date: 2002-12-03 19:01:10 UTC
 meas_id: 4 items (dict)
 nchan: 376
 projs: PCA-v1: off, PCA-v2: off, PCA-v3: off, Average EEG reference: off
 sfreq: 150.2 Hz
>
```

When I then calculate the covariance matrix by calling `Cov = mne.compute_covariance(epochs, return_estimators=True)`, I get :

```
Loading data for 72 events and 106 original time points ...
    Rejecting  epoch based on EOG : ['EOG 061']
    Rejecting  epoch based on EOG : ['EOG 061']
    Rejecting  epoch based on EOG : ['EOG 061']
    Rejecting  epoch based on EOG : ['EOG 061']
    Rejecting  epoch based on EOG : ['EOG 061']
    Rejecting  epoch based on MAG : ['MEG 1711']
    Rejecting  epoch based on EOG : ['EOG 061']
    Rejecting  epoch based on EOG : ['EOG 061']
    Rejecting  epoch based on EOG : ['EOG 061']
    Rejecting  epoch based on EOG : ['EOG 061']
    Rejecting  epoch based on EOG : ['EOG 061']
    Rejecting  epoch based on EOG : ['EOG 061']
    Rejecting  epoch based on EOG : ['EOG 061']
    Rejecting  epoch based on EOG : ['EOG 061']
    Rejecting  epoch based on EOG : ['EOG 061']
    Rejecting  epoch based on EOG : ['EOG 061']
    Rejecting  epoch based on EOG : ['EOG 061']
17 bad epochs dropped
Loading data for 73 events and 106 original time points ...
    Rejecting  epoch based on EOG : ['EOG 061']
    Rejecting  epoch based on EOG : ['EOG 061']
    Rejecting  epoch based on EOG : ['EOG 061']
    Rejecting  epoch based on EOG : ['EOG 061']
    Rejecting  epoch based on MAG : ['MEG 1711']
    Rejecting  epoch based on EOG : ['EOG 061']
    Rejecting  epoch based on EOG : ['EOG 061']
    Rejecting  epoch based on EOG : ['EOG 061']
    Rejecting  epoch based on EOG : ['EOG 061']
    Rejecting  epoch based on EOG : ['EOG 061']
    Rejecting  epoch based on EOG : ['EOG 061']
    Rejecting  epoch based on EOG : ['EOG 061']
12 bad epochs dropped
Loading data for 73 events and 106 original time points ...
    Rejecting  epoch based on EOG : ['EOG 061']
    Rejecting  epoch based on EOG : ['EOG 061']
    Rejecting  epoch based on EOG : ['EOG 061']
    Rejecting  epoch based on EOG : ['EOG 061']
    Rejecting  epoch based on EOG : ['EOG 061']
    Rejecting  epoch based on EOG : ['EOG 061']
6 bad epochs dropped
Loading data for 70 events and 106 original time points ...
    Rejecting  epoch based on EOG : ['EOG 061']
    Rejecting  epoch based on EOG : ['EOG 061']
    Rejecting  epoch based on EOG : ['EOG 061']
    Rejecting  epoch based on EOG : ['EOG 061']
    Rejecting  epoch based on EOG : ['EOG 061']
    Rejecting  epoch based on EOG : ['EOG 061']
    Rejecting  epoch based on EOG : ['EOG 061']
    Rejecting  epoch based on EOG : ['EOG 061']
    Rejecting  epoch based on EOG : ['EOG 061']
    Rejecting  epoch based on EOG : ['EOG 061']
    Rejecting  epoch based on EOG : ['EOG 061']
    Rejecting  epoch based on EOG : ['EOG 061']
    Rejecting  epoch based on EOG : ['EOG 061']
    Rejecting  epoch based on EOG : ['EOG 061']
14 bad epochs dropped
Loading data for 15 events and 106 original time points ...
0 bad epochs dropped
Loading data for 16 events and 106 original time points ...
    Rejecting  epoch based on EOG : ['EOG 061']
    Rejecting  epoch based on EOG : ['EOG 061']
2 bad epochs dropped
Computing rank from data with rank=None
    Using tolerance 1.4e-08 (2.2e-16 eps * 305 dim * 2.1e+05  max singular value)
    Estimated rank (mag + grad): 302
    MEG: rank 302 computed from 305 data channels with 3 projectors
    Created an SSP operator (subspace dimension = 3)
    Setting small MEG eigenvalues to zero (without PCA)
Reducing data rank from 305 -> 302
Estimating covariance using EMPIRICAL
Done.
Number of samples used : 28408
[done]
```

So now `Cov` stores the mne covariance matrix object, and `np.shape(Cov['data'])` tells me that it is a 305 x 305 numpy array. 
As far as I understand, what is happening is that it "drops" the epochs that contribute to noise, resulting in a reduced num_epochs x num_epochs covariance matrix (num_epochs is reduced to 305 in this case). I have two questions about this: 

**1. Why is Cov['data'] of shape num_epochs x num_epochs (or 305 x 305) and not num_epochs x num_channels x num_channels (or 305 x 376 x 376)? 
2. Is it possible to get a list of the epochs that were dropped? Or to get a list of the non-dropped epochs that contribute to the reduced covariance matrix?**

Many thanks!

agramfort, 2021-05-21:
a data covariance is channel x channel

it's 305x305 as you have 305 good MEG channels

ALex

>

drammock, 2021-05-21:
covariance dimension generally starts out as `n_channels x n_channels` and gets reduced from there depending on the rank of the data.  In your case it looks like you probably dropped the EEG channels at some point before computing covariance; compare an excerpt of my output to what yours above looks like:
```
omputing rank from data with rank=None
    Using tolerance 2.6e-08 (2.2e-16 eps * 305 dim * 3.8e+05  max singular value)
    Estimated rank (mag + grad): 302
    MEG: rank 302 computed from 305 data channels with 3 projectors
    Using tolerance 3.5e-10 (2.2e-16 eps * 59 dim * 2.7e+04  max singular value)
    Estimated rank (eeg): 59        <---------- MISSING IN YOUR OUTPUT
    EEG: rank 59 computed from 59 data channels with 0 projectors        <---------- MISSING IN YOUR OUTPUT
    Created an SSP operator (subspace dimension = 3)
    Setting small MEG eigenvalues to zero (without PCA)
    Setting small EEG eigenvalues to zero (without PCA)        <---------- MISSING IN YOUR OUTPUT
```

ilonadem, 2021-05-21:
I see -- but then why does the number of epochs have no effect on the dimensions of the covariance matrix produced?

drammock, 2021-05-21:
> I see -- but then why does the number of epochs have no effect on the dimensions of the covariance matrix produced?

that's just not how the function is designed to work.  You could get a separate covariance estimate for each epoch by using a for-loop if you want.  But estimating covariance based on a small number of time samples will be less accurate, so the estimates you get from each epoch may not agree very well.  Bear in mind that the epochs passed into the function are (or should be!) from the same subject, same recording hardware, same recording session, etc.  The sensor covariance *ought* to be approximately stable over the course of a recording, so by incorporating info from all epochs into the covariance estimate, we are more likely to estimate the true covariance accurately.

ilonadem, 2021-05-21:
Thank you so much, that makes a lot of sense!

closed 2021-05-21 by ilonadem