mne-tools/mne-python #9421: GSoC 2021 - Refactoring mne.connectivity into men-connectivity repository

adam2392, 2021-05-21:
This issue is meant to track the progress of refactoring existing functionality inside mne-python's "connectivity" sub-module into an entirely new repository.

- [ ] Create new repository, `mne-connectivity`, inside mne.tools (I think I have access)
- [ ] Get CI and basic GH action tools setup, based on `mne-bids` and `mne-python` CI
- [ ] Refactor existing code into the `mne-connectivity` repo and include a validation notebook that validates that code in this refactored setting matches what is here

To track detailed progress inside `mne-connectivity`, such as getting CI up and running, I'll point to issues in that specific repository.

## Higher level goals

I think @agramfort spoke about this a bit with me and I agree: there is really no "great way" of saying which algorithm or model is better in certain instances. There are many "proposed connectivity algorithms", whether they are time-domain, or free-domain based. So the metric moving forward in this new repo is to at least standardize what you should expect on a dataset.

Given 1-3 datasets, running a specific algorithm, one should expect certain performance or values. The current goal is to leverage openneuro and pull datasets for doing this in CI.

## Reference Issues

#9060 #9030 #8729

larsoner, 2021-05-21:
@adam2392 I read this only after making my comment in the other repo, which seems to echo almost 1:1 what you wrote here -- so sounds good to me :)

> The current goal is to leverage openneuro and pull datasets for doing this in CI.

I would start smaller and try using toy or simulated datasets -- preferably there are some of these in the reference papers. Once that works, examples that build on CircleCI to give outputs for real-data inputs are great, but I wouldn't start from these if possible because debugging them will be harder both in terms of eyeballing errors and in terms of CI time (they are going to require a big download and a long processing time compared to a toy example).

adam2392, 2021-06-24:
@drammock @larsoner and @britta-wstnr and I were discussing today and they suggested I ask you:

What do we do with all the mne.connectivity examples inside mne-python? 
- should we have a link directly to the mne-connectivity package?
- what’s the best way to advertise this?

Any preferences there?

drammock, 2021-06-24:
> What do we do with all the mne.connectivity examples inside mne-python?

I can think of 2 broad options:

1. move the examples to the doc site of `mne-connectivity`
2. leave them on the MNE-Python site, but import the needed functions from `mne-connectivity`

I think of those I prefer option 1.  Each examples category already includes a short text description, so it would be easy to edit that to say something like "connectivity analysis algorithms have been moved to a separate package, mne-connectivity.  See the examples there." (and include a link).

larsoner, 2021-06-24:
Agreed, my inclination is to have the "Connectivity" section of the examples now just be a single sentence pointing to the mne-connectivity examples page. And the when you click on the front page icon that advertises connectivity, it should take you to the mne-connectivity examples page.

britta-wstnr, 2021-06-24:
Yes, that was also my thought, +1 for option 1. I think it is important though that connectivity analysis in some ways stays advertised on the main MNE-Python homepage, as it is something many users look for.

wmvanvliet, 2021-07-08:
This sounds like something I could merge the [conpy](https://github.com/aaltoimaginglanguage/conpy) module into.

PR #9493 (2021-08-17): [MRG, API] Initiate deprecation of mne.connectivity submodule
#### Reference issue
Fixes: #9421 

#### What does this implement/fix?
Initiates the deprecation of mne.connectivity into `mne-connectivity` 

#### Additional information
TODO:

- [x] figure out how to do CI releases on men-connectivity because I can't do a release manually. M1 Macs and even my Linux computer stalling when building the docs locally with the graphs. Not sure why. -> Make v0.1 release which is basically the same as what men-python.connectivity has rn

Afterwards, all changes to connectivity will occur in this new repo [... cut]
closed 2021-08-17 by larsoner