mne-tools/mne-python #9425: save_movie does not seem to interpret arguments

timpx, 2021-05-25:
#### Describe the bug
save_movie function does not seem to interpret arguments. For instance, with the time_viewer argument set to True, the saved movie does not have a visible time viewer. The time_dilatation argument does not seem to have any impact either. Etc.

#### Steps to reproduce
Run the [sEEG tutorial](https://mne.tools/stable/auto_tutorials/clinical/20_seeg.html) and uncomment the following line:
```python
brain.save_movie(time_dilation=3, interpolation='linear', framerate=10,
                  time_viewer=True, filename='./mne-test-seeg.m4')
```

#### Expected results
I would expect the arguments to work. Especially time_dilatation, because right now for me it takes forever to save the movie.

#### Actual results
Resulting movie is 1 min 40 sec without time courses, unlike what is shown on the sEEG tutorial webpage.

#### Additional information
Platform:      Linux-5.4.0-72-generic-x86_64-with-glibc2.31
Python:        3.9.2 | packaged by conda-forge | (default, Feb 21 2021, 05:02:46)  [GCC 9.3.0]
Executable:    ~/anaconda3/envs/neo_mne/bin/python
CPU:           x86_64: 24 cores
Memory:        250.2 GB

mne:           0.23.0
numpy:         1.19.2 {blas=openblas, lapack=openblas}
scipy:         1.5.3
matplotlib:    3.3.4 {backend=Qt5Agg}

sklearn:       0.23.2
numba:         0.53.1
nibabel:       3.2.1
nilearn:       0.7.1
dipy:          1.2.0
cupy:          Not found
pandas:        1.2.4
mayavi:        4.7.2
pyvista:       0.29.1 {pyvistaqt=0.3.0, OpenGL 4.5.0 NVIDIA 390.143 via Quadro P5000/PCIe/SSE2}
vtk:           9.0.1
PyQt5:         5.12.3

larsoner, 2021-05-25:
@timpx are you running this in a Jupyter notebook where the viewer is embedded in a webpage, or plain Python (or IPython) interpreter where the viewer pops up as a Qt5 application window?

timpx, 2021-05-25:
I tried both, for both the pyvista 3d backend is used, and the plot_3d pops up a window. Same issue for both though.

GuillaumeFavelier, 2021-05-25:
Do you mind updating `pyvista` and `pyvistaqt` to the latest version (`0.30.1` and `0.4.0` respectively) and try again?

GuillaumeFavelier, 2021-05-25:
By inspecting `_save_movie_tv` I indeed found inconsistencies with input parameters, I'll open a PR soon to fix this.

timpx, 2021-05-25:
Thanks! Updating pyvista and pyvistaqt did not help indeed.

closed 2021-05-27 by larsoner