mne-tools/mne-python #9428: Info rank computation fails if picking only 'grad' or 'mag'

agramfort, 2021-05-27:
try to run this:

```python
import os.path as op
import mne
from mne.datasets import somato

data_path = somato.data_path()
subject = '01'
task = 'somato'
raw_fname = op.join(data_path, 'sub-{}'.format(subject), 'meg',
                    'sub-{}_task-{}_meg.fif'.format(subject, task))

raw = mne.io.read_raw_fif(raw_fname)
events = mne.find_events(raw, stim_channel='STI 014')
picks = mne.pick_types(raw.info, meg=True, eeg=False, eog=True, stim=False)
event_id, tmin, tmax = 1, -1., 3.
baseline = (None, 0)
epochs = mne.Epochs(raw, events, event_id, tmin, tmax, picks=picks,
                    baseline=baseline, reject=dict(grad=4000e-13, eog=350e-6),
                    preload=True)
evoked = epochs.average()

print('Rank with all MEG')
print(mne.compute_rank(evoked, 'info'))

print('Rank with only GRAD')
print(mne.compute_rank(evoked.copy().pick('grad'), 'info'))
```

you will get:

```
Rank with all MEG
{'meg': 64}
Rank with only GRAD
{'grad': 204}
```

it shows you that if you pick the grad it ignores the rank computation derived from maxfilter.

@larsoner thoughts?

thx @PABannier for spotting the bug...

larsoner, 2021-05-27:
Agreed this seems like a bug

agramfort, 2021-05-28:
do you assign this to you?

>

larsoner, 2021-05-28:
Yep, should be able to take a look next week

PR #9435 (2021-06-01): FIX : rank computation from info now uses SSS proc history if only gr…
…ad or mag are present

closes #9428
closed 2021-06-01 by hoechenberger