mne-tools/mne-python #9429: NotImplementedError for larger files in read_raw_nihon()

PhlipPhlops, 2021-05-27:
https://github.com/mne-tools/mne-python/blob/3b69e9b936ee01fcb357e732ee35cb75086a6bf8/mne/io/nihon/nihon.py#L214

Trying to read larger Nihon Koden EEG files, I've been hitting the error raised at this line above.
Any bead on when this function might be implemented? If there are no designs I'm happy to try my hand at it if someone's willing to give me a bit of direction

agramfort, 2021-05-27:
ping @fraimondo

fraimondo, 2021-05-28:
Which one of the two errors was?

This is not yet implemented because I did not have any large files to test. I will need a large file that raises the `NotImplementedError`, as well as the same large file but in EDF format to compare.

PhlipPhlops, 2021-05-28:
It was the Control Block error (first one of the two)
I have a few anonymized .EEG files I can let you use to test with; they all run into this same error. I'll convert a few into EDF for you.
If you send me an email at phillkin@gmail.com I can send you a google drive link to the folder

fraimondo, 2021-05-28:
> If you send me an email at ***@***.*** ***@***.***> I can send you a google drive link to the folder

Send them to ***@***.*** ***@***.***>

I’ll have a look.

PhlipPhlops, 2021-05-28:
Thanks Fede, the email you sent is censored so I sent the link to the email I found in your github profile. Let me know if you got them

fraimondo, 2021-06-01:
Got them and developed a solution. However, I can't test it yet.  These are the files I got:

```
87M FA0750X2.EEG
87M FA0750X2_1-1.edf
107M FA0751EA_2013.EEG
32M FA0751EA_2013_1-1.edf
48M FA07548P.EEG
47M FA0755PV.EEG
```

1. `FA0750X2.EEG` has only one datablock, so it worked before the modification
2. `FA0751EA_2013.EEG` has two datablocks, but the EDF export is split by datablock and `FA0751EA_2013_1-1.edf` is the first one (check the size)
3. `FA07548P.EEG` and `FA0755PV.EEG` have two or more datablocks, but I don't have the EDF files.

PhlipPhlops, 2021-06-02:
Love to hear it :heart: 
I've dropped in the drive the two .edf datablocks for `FA07548P.EEG` and the four .edf datablocks for `FA0755PV.EEG` for your testing pleasure (sorry it took a minute, my partner with the files was too sick to move yesterday)

fraimondo, 2021-06-03:
I have a couple more of questions: 

1- Are you still able to access the system to do a couple of recordings for the testing dataset?
2- Why are this recordings split into several datablocks? Is the recording paused in the middle?
3- If 1 and 2 are positive, can you generate a recording that is just a few seconds but with many pauses in between? With this testing data we can integrate it in the MNE testing data and perform automated tests. The electrodes don't need to be connected.

PhlipPhlops, 2021-06-04:
We're not confident about #1, but my partner is heading to the hospital today to see if she can capture some recordings like you described for these automated tests. If you have any ideas you'd like to test, please let me know soon so she may do what she can while she's at the hospital today.

It seems you are right, that pausing is what is separating these recordings into datablocks, but we are not confident just yet. Will see later today

Again if there are any ideas you'd like to test, please let me know soon! We have limited access to the EEG system

PhlipPhlops, 2021-06-04:
She found a .EEG file confirmed with no pauses, and when converting to .edf it only produces one .edf file. That doesn't confirm that pauses are what's separating the datablocks, but it's evidence.

PhlipPhlops, 2021-06-05:
Looks like we might not have access to generate new files for a while. Are you able to use any of the files in the drive for automated testing?

agramfort, 2021-06-06:
for automated testing we cannot use big files. I guess here if the code is
tested once and written so that it is covered
by our test it's already a clear progress.

>

fraimondo, 2021-06-07:
I just pushed to #9437 

The PR works on the files sent.

PhlipPhlops, 2021-06-18:
Apologies for the delay @fraimondo , thank you so much for your work, this solves our problem beautifully

closed 2021-06-18 by PhlipPhlops