mne-tools/mne-python #9433: ENH: Website API class-methods in "On this page"(right dock)

marsipu, 2021-05-30:
## Describe the new feature or enhancement
For the documentation of classes in the [API on the Website](https://mne.tools/dev/python_reference.html) it may be easier to navigate if their methods were outlined in the right sidebar ("On this page").

[image]

## Implementation
I am not sure how to implement this. Maybe some change of the configuration of sphinx.autodoc? Or would it need changes on how the pydata-sphinx-theme handles the API?

agramfort, 2021-05-30:
do you know a project that does this? so we can copy the way they do it...

>

marsipu, 2021-05-30:
> 
> do you know a project that does this? so we can copy the way they do it...
> […](#)

I checked all the projects using the pydata-sphinx-theme (from the [list in README](https://github.com/pydata/pydata-sphinx-theme)). Unfortunately I found none which documented its classes that way. So it is probably something which had to be implemented on the side of pydata-sphinx-theme?

agramfort, 2021-05-31:
yes it would then require an improvement in pydata-sphinx-theme
you can open an issue on their repo to get feedback maybe

>

drammock, 2021-05-31:
This may not be very easy I suspect.  The page elements that end up shown under "on this page" are `h2`, `h3` etc.  The class methods are `dt` elements.  This is the function that determines the in-page TOC:

https://github.com/pydata/pydata-sphinx-theme/blob/3cdae50ef8cd7b1967e69c3b2e337a7c8a453f83/pydata_sphinx_theme/__init__.py#L137

However, if you *could* fix this, it would also mean that the glossary page entries would end up in the "on this page" menu (which would be good, and which I've asked them about before).

marsipu, 2021-05-31:
Thank you for the hint, I'll ask the pydata-developers again and try to learn a bit more about it.