mne-tools/mne-python #9441: Plot all channels with raw.plot_psd

cbrnr, 2021-06-02:
We have already discussed and fixed the problem that it was impossible to plot the PSD of bad channels (#6637, #6678). However, I find the solution still not ideal. I frequently need to plot PSDs of all channels, and the only way to do that is to pass all channel indices or names to `picks`. I'd prefer a string value, but unfortunately `"all"` and `"data"` are already taken and just include good channels. There's also the default third option (`None`), which seems to be equal to `"all"`.

Does anyone have a good suggestion how to solve this issue?

agramfort, 2021-06-04:
I would do it such that:

raw.plot_psd(spatial_colors=False, exclude=None)

plots all channels including bads. Woudl you also plot simple with different line style or color?

cbrnr, 2021-06-04:
> I would do it such that:
>
> `raw.plot_psd(spatial_colors=False, exclude=None)`

I guess you mean `raw.plot_psd(spatial_colors=False, picks=None)`? Yes, this would work, but then we need to change the default to `picks="all"` in order not to break current behavior.

> Woudl you also plot simple with different line style or color?

Not sure what you mean, but usually I use the default `spatial_colors=True` but with `average=False` (otherwise I'd get an average over all channels).

agramfort, 2021-06-04:
>
> I guess you mean raw.plot_psd(spatial_colors=False, picks=None)? Yes,
> this would work, but then we need to change the default to picks="all" in
> order not to break current behavior.
>

presently the default is picks=None and this plots all good channels. I
propose to add an exclude param
that would default to 'bads' and that would not change the default
behavior. So when you change
the default with

 raw.plot_psd(spatial_colors=False, picks=None, exclude=[])

it would plot the bads. This is consistent with evoked.plot for example.

> Woudl you also plot simple with different line style or color?
>
> Not sure what you mean, but usually I use the default spatial_colors=True
> but with average=False (otherwise I'd get an average over all channels).
>

I would copy the behavior of evoked.plot (colors when exclude is [])

makes sense?

cbrnr, 2021-06-09:
Sounds good. Any objections? Otherwise I'll make a PR.

larsoner, 2021-06-09:
Sounds reasonable to me

drammock, 2021-06-09:
+1 from me anytime we're make API + behavior more consistent between different plotting functions :)

PR #9519 (2021-07-07): Add exclude parameter to raw.plot_psd()
Fixes #9441.

This adds an `exclude` parameter to `raw.plot_psd()`, which makes it possible to plot the PSD of *all* channels (including bads) using `raw.plot_psd(exclude=[])`.

Questions/To Do:
- [x] Add changelog in API or new section?
- [x] Add `.. versionadded :: 0.24.0` tag?
- [x] Add note somewhere (docstring?) to show how to plot the PSD for all channels (including bads)?
closed 2021-07-07 by drammock