mne-tools/mne-python #9451: Recommended way to launch the coregistration GUI

hoechenberger, 2021-06-08:
Currently, the [`mne.gui.coregistration()` docs](https://mne.tools/stable/generated/mne.gui.coregistration.html) say:

```
The recommended way to use the GUI is through bash with:

$ mne coreg
```

- Why is that "recommended"? Working with "normal" users I got the impression that having to use the command line makes it much more inefficient (if not difficult) for them to loop over multiple participants, which is often what is needed, than when using Python code.
- Why does it specifically recommended to use `bash`? I'm using `zsh`, which is now also the default on macOS.

### Proposal

We should mention that **an alternative way** to start the coreg gui is via the CLI, but I'd remove the implications that it's "recommended" and requires bash.

mmagnuski, 2021-07-05:
I don't know if this is related but using mne coreg recently I always end up with an unresponsive kernel once I close the GUI (Windows). IIRC restarting the spyder console does not always help and I have to restart Spyder. Maybe that's why it is recommended to start coreg from the console.
I can test more if that would be helpful.

alexrockhill, 2022-06-17:
From https://github.com/mne-tools/mne-python/issues/10779, https://mne.tools/dev/auto_tutorials/forward/20_source_alignment.html#defining-the-headmri-trans-using-the-gui looks like it has the space for code to launch the `coreg` GUI but it's not actually executed. I think it should be executed here and maybe we can add a video tutorial since that's the new thing.

closed 2022-06-24 by larsoner