mne-tools/mne-python #9466: Add norm parameter to plot_topomap

cbrnr, 2021-06-11:
I would like to plot ERD/ERS values on a topomap using `mne.viz.plot_topomap`. I can already select a suitable diverging colormap such as `RdBu_r` (red&ndash;white&ndash;blue), but my values are not symmetrical around 0 (they range from e.g. &ndash;1 to +2.5). Because I would like to map white to zero, I need to set a different [colormap normalization](https://matplotlib.org/stable/tutorials/colors/colormapnorms.html) such as the [TwoSlopeNorm](https://matplotlib.org/stable/tutorials/colors/colormapnorms.html#twoslopenorm-different-mapping-on-either-side-of-a-center). The `mne.viz.plot_topomap` function uses [imshow](https://matplotlib.org/stable/api/_as_gen/matplotlib.pyplot.imshow.html) internally, which already has a `norm` parameter. Therefore, we could pretty easily expose this new parameter in `plot_topomap`. I'd have to test if this really works for my use case, but in principle this seems like a promising approach. Does anyone have any objections or alternative suggestions? Otherwise I'll submit a PR.

cbrnr, 2021-06-11:
Some additional information: here's code to produce a topomap with values ranging from -1 to 2.5:

```python
import numpy as np
import matplotlib.pyplot as plt
from matplotlib.cm import get_cmap
import mne
from mne.viz import plot_topomap

rng = np.random.default_rng(seed=42)
v = rng.uniform(low=-1, high=2.5, size=64)
v[:3] = [-1, 0, 2.5]

montage = mne.channels.make_standard_montage("biosemi64")
info = mne.create_info(montage.ch_names, 256, "eeg").set_montage("biosemi64")
cmap = get_cmap("RdBu_r")

fig, ax = plt.subplots()
im = plot_topomap(v, info, vmin=-1, vmax=2.5, cmap=cmap, axes=ax, show=False)
cb = fig.colorbar(im[0], ax=ax)
```
[image]

Notice how white does not map to 0, but to 0.75 (the middle of the data range).

By adding a `norm` parameter, this example would look as follows:

```python
import numpy as np
import matplotlib.pyplot as plt
from matplotlib.cm import get_cmap
from matplotlib.colors import TwoSlopeNorm
import mne
from mne.viz import plot_topomap

rng = np.random.default_rng(seed=42)
v = rng.uniform(low=-1, high=2.5, size=64)
v[:3] = [-1, 0, 2.5]

montage = mne.channels.make_standard_montage("biosemi64")
info = mne.create_info(montage.ch_names, 256, "eeg").set_montage("biosemi64")
cmap = get_cmap("RdBu_r")
norm = TwoSlopeNorm(vmin=-1, vcenter=0, vmax=2.5)

fig, ax = plt.subplots()
im = plot_topomap(v, info, vmin=-1, vmax=2.5, cmap=cmap, axes=ax, norm=None, show=False)
cb = fig.colorbar(im[0], ax=ax)
```

[image]

Now white correctly maps to 0.

drammock, 2021-06-11:
the only alternate suggestion I have is to expose `clim` instead, and you could pass something like `clim=np.array([-1, 1]) * max(abs(data))`. That would put zero in the middle (white) but in your example the blues would be not nearly so intense.

cbrnr, 2021-06-11:
Yes, but that's what I actually wanted to avoid. I really need to show the full color range for my asymmetrical data.

drammock, 2021-06-11:
+1 for a PR, then 🙂

PR #9468 (2021-06-24): Add color normalization parameter to plot_topomap
Fixes #9466. Things to do/discuss:

- I've named the parameter `cnorm` (for "color normalization"), because I think this better describes its purpose. It also avoids a name clash with an already existing local name `norm` (which has something to do with the L2 norm, not entirely sure). However, Matplotlib calls this parameter `norm`, so if we want to mimic their naming scheme we could also call our parameter `norm` instead.
- I need to add a test.
- When passing `cnorm`, simultaneously passing `vmin` and `vmax` is deprectated, because these are [... cut]
closed 2021-06-24 by larsoner