mne-tools/mne-python #9469: mne.filter.filter_data : All picks must be < n_channels (19), got 19

0anurag0, 2021-06-12:
#### Describe the bug
I get this value while trying to get a high pass filter
ValueError: All picks must be < n_channels (19), got 19

#### Steps to reproduce
fname = mne.datasets.sample.data_path() + '/MEG/sample/sample_audvis_raw.fif'
raw = mne.io.read_raw_fif(fname)
raw.info
mne.filter.filter_data(raw,h_freq=0.1,l_freq=None,sfreq=600.61)

[MWE]: https://en.wikipedia.org/wiki/Minimal_Working_Example
[built-in datasets]: https://mne.tools/dev/overview/datasets_index.html
[sample]: https://mne.tools/dev/overview/datasets_index.html#sample

#### Expected results
a successful execution of high pass filter

#### Actual results
ValueError: All picks must be < n_channels (376), got 376

#### Additional information
I have tried this with my dataset too, same error
I have tried using picks too like below:
picks=['Fp1',"Fp2","F3","F4","Fz","T5","T6","T3","T4","C3","C4","Cz"]
or 
picks=[ 0  1  2  3  4  5  6  7  8  9 10 11 12 17]
or picks=[0,1,2,3,4,5,6,7,8,9,10,11,12,17]
but same error

Platform:      Linux-5.4.104+-x86_64-with-Ubuntu-18.04-bionic
Python:        3.7.10 (default, May  3 2021, 02:48:31)  [GCC 7.5.0]
Executable:    /usr/bin/python3
CPU:           x86_64: 2 cores
Memory:        12.7 GB

mne:           0.23.0
numpy:         1.19.5 {blas=openblas, lapack=openblas}
scipy:         1.4.1
matplotlib:    3.2.2 {backend=module://ipykernel.pylab.backend_inline}

sklearn:       0.22.2.post1
numba:         0.51.2
nibabel:       3.0.2
nilearn:       Not found
dipy:          Not found
cupy:          Not found
pandas:        1.1.5
mayavi:        Not found
pyvista:       Not found
vtk:           Not found

cbrnr, 2021-06-12:
There are several issue with your approach:

1. `mne.filter.filter_data` expects an `np.array` as its `data` argument, not a `raw` object. Use the `filter` method instead, i.e. `raw.filter()`.
2. If you want to highpass-filter the data, you need to specify the `l_freq` argument (lower edge frequency); set `h_freq` to `None` if you don't want a lowpass filter.
3. Data needs to be loaded for the filter operation to work; either pass `preload=True` when loading the data, or run `raw.load_data()` right before filtering.

0anurag0, 2021-06-13:
Thanks for Clearing the doubt!
I tried all the changes and it works completely fine.

closed 2021-06-13 by 0anurag0