mne-tools/mne-python #9470: creating epoches without event

0anurag0, 2021-06-13:
#### Describe the bug
trying to create an epoch with a continuous dataset which doesn't have events so creating events manually but unsuccessful.
using np array yet it says needs numpy array of int and has numpy.ndarray

#### Steps to reproduce
events=np.linspace(0,15*60,600))
epochs = mne.Epochs(edf_h01_r_hf.,events,  tmin=-0.2, tmax=0.5, preload=True)

#### Expected results
epoch formation

#### Actual results
TypeError: events should be a NumPy array of integers, got <class 'numpy.ndarray'>

#### Additional information
Platform:      Linux-5.4.0-74-generic-x86_64-with-glibc2.29
Python:        3.8.5 (default, May 27 2021, 13:30:53)  [GCC 9.3.0]
Executable:    /usr/bin/python3
CPU:           x86_64: 4 cores
Memory:        3.7 GB

mne:           0.23.0
numpy:         1.19.5 {blas=openblas, lapack=openblas}
scipy:         1.6.3
matplotlib:    3.4.2 {backend=module://ipykernel.pylab.backend_inline}

sklearn:       0.24.2
numba:         Not found
nibabel:       Not found
nilearn:       Not found
dipy:          Not found
cupy:          Not found
pandas:        1.2.4
mayavi:        Not found
pyvista:       Not found
vtk:           Not found

agramfort, 2021-06-13:
can you see if this helps?

https://mne.tools/dev/auto_tutorials/epochs/60_make_fixed_length_epochs.html
https://mne.tools/dev/generated/mne.make_fixed_length_epochs.html#mne.make_fixed_length_epochs
https://mne.tools/dev/generated/mne.make_fixed_length_events.html

please use the forum https://mne.discourse.group/ to ask usage questions.

close the issue if your problem is solved. thanks

0anurag0, 2021-06-13:
Thanks! that was  a huge help.
I will remember to ask in the forum next time.

closed 2021-06-13 by 0anurag0