mne-tools/mne-python #9473: Plotting 3D Backend Not Working - Mac M1

adam2392, 2021-06-14:
#### Describe the bug
I'm trying to run the `mne_inverse_envelope_correlation.py` example locally on my Mac M1 13 inch, 2021, but get a `RuntimeError: Could not load any valid 3D backend: pyvista, mayavi, notebook`.

#### Expected results
I would expect that the `conda install pyvista` would make this work?

#### Actual results
```
Traceback (most recent call last):
  File "/Users/adam2392/Documents/mne-connectivity/examples/mne_inverse_envelope_correlation.py", line 100, in <module>
    brain = stc.plot(
  File "/Users/adam2392/miniforge3/envs/mne/lib/python3.8/site-packages/mne/source_estimate.py", line 651, in plot
    brain = plot_source_estimates(
  File "<decorator-gen-153>", line 24, in plot_source_estimates
  File "/Users/adam2392/miniforge3/envs/mne/lib/python3.8/site-packages/mne/viz/_3d.py", line 1765, in plot_source_estimates
    backend = _get_3d_backend()
  File "/Users/adam2392/miniforge3/envs/mne/lib/python3.8/site-packages/mne/viz/backends/renderer.py", line 153, in _get_3d_backend
    raise RuntimeError(f'Could not load any valid 3D backend: '
RuntimeError: Could not load any valid 3D backend: pyvista, mayavi, notebook
(mne) adam2392@adams-mbp-2 mne-connectivity % pip install pyvista
Requirement already satisfied: pyvista in /Users/adam2392/miniforge3/envs/mne/lib/python3.8/site-packages (0.31.1)
Requirement already satisfied: pillow in /Users/adam2392/miniforge3/envs/mne/lib/python3.8/site-packages (from pyvista) (8.2.0)
Requirement already satisfied: numpy in /Users/adam2392/miniforge3/envs/mne/lib/python3.8/site-packages (from pyvista) (1.20.3)
Requirement already satisfied: scooby>=0.5.1 in /Users/adam2392/miniforge3/envs/mne/lib/python3.8/site-packages (from pyvista) (0.5.7)
Requirement already satisfied: imageio in /Users/adam2392/miniforge3/envs/mne/lib/python3.8/site-packages (from pyvista) (2.9.0)
Requirement already satisfied: transforms3d==0.3.1 in /Users/adam2392/miniforge3/envs/mne/lib/python3.8/site-packages (from pyvista) (0.3.1)
Requirement already satisfied: meshio<5.0,>=4.0.3 in /Users/adam2392/miniforge3/envs/mne/lib/python3.8/site-packages (from pyvista) (4.4.6)
Requirement already satisfied: vtk in /Users/adam2392/miniforge3/envs/mne/lib/python3.8/site-packages (from pyvista) (9.0.1)
Requirement already satisfied: appdirs in /Users/adam2392/miniforge3/envs/mne/lib/python3.8/site-packages (from pyvista) (1.4.4)
```

#### Additional information
```
>>> mne.sys_info()
Platform:      macOS-11.4-arm64-arm-64bit
Python:        3.8.10 | packaged by conda-forge | (default, May 11 2021, 06:27:18)  [Clang 11.1.0 ]
Executable:    /Users/adam2392/miniforge3/envs/mne/bin/python
CPU:           arm: 8 cores
Memory:        16.0 GB

mne:           0.23.0
numpy:         1.20.3 {blas=NO_ATLAS_INFO, lapack=lapack}
scipy:         1.6.3
matplotlib:    3.4.2 {backend=MacOSX}

sklearn:       0.24.2
numba:         Not found
nibabel:       Not found
nilearn:       Not found
dipy:          Not found
cupy:          Not found
pandas:        1.2.4
mayavi:        Not found
pyvista:       0.31.1 {OpenGL 4.1 Metal - 71.6.4 via Apple M1}
vtk:           9.0.1
PyQt5:         Not found
```

agramfort, 2021-06-14:
@PABannier @dengemann can you tell @adam2392 what 

`mne sys_info` returns on your M1?

PABannier, 2021-06-15:
@adam2392 Hey Adam! Find below the output of `mne sys_info` on my M1.

```
Platform:      macOS-11.2.2-x86_64-i386-64bit
Python:        3.9.2 | packaged by conda-forge | (default, Feb 21 2021, 05:02:20)  [Clang 11.0.1 ]
Executable:    /Users/pierre-antoine/anaconda3/envs/mne/bin/python3.9
CPU:           i386: 8 cores
Memory:        16.0 GB

mne:           0.24.dev0
numpy:         1.20.2 {blas=NO_ATLAS_INFO, lapack=lapack}
scipy:         1.6.2
matplotlib:    3.4.1 {backend=MacOSX}

sklearn:       0.24.1
numba:         0.53.1
nibabel:       3.2.1
nilearn:       0.7.1
dipy:          1.4.0
cupy:          Not found
pandas:        1.2.3
mayavi:        4.7.2
pyvista:       0.29.0 {pyvistaqt=0.3.0, OpenGL 4.1 Metal - 71.0.7 via Apple M1}
vtk:           9.0.1
PyQt5:         5.12.3
```

adam2392, 2021-06-15:
@PABannier may I ask how you install pyqt5? I tried:

`conda install -c anaconda pyqt` but it failed on my:

```
Current channels:

  - https://conda.anaconda.org/anaconda/osx-arm64
  - https://conda.anaconda.org/anaconda/noarch
  - https://conda.anaconda.org/conda-forge/osx-arm64
  - https://conda.anaconda.org/conda-forge/noarch
```

larsoner, 2021-06-15:
@adam2392 they are not using the native `arm64` but rather the `i386` version. If you use this version of anaconda you should be able to get PyQt5 (and other packages that don't yet natively support `arm64`)

adam2392, 2021-06-15:
Stupid question, does that mean use the native installers here: https://docs.conda.io/en/latest/miniconda.html#macosx-installers?

For reference: I used `miniforge`, which was an open-source project deved for the sake of arm64.

adam2392, 2021-06-16:
I just tried it on my Linux workstation and had the same issue. Asking here in case it's a bug with `_get_3d_backend`.

```
WARNING: /home/adam2392/Documents/mne-connectivity/examples/mne_inverse_coherence_epochs.py failed to execute correctly: Traceback (most recent call last):
  File "/home/adam2392/Documents/mne-connectivity/examples/mne_inverse_coherence_epochs.py", line 141, in <module>
    brain = coh_stc.plot('sample', 'inflated', 'both',
  File "/home/adam2392/miniconda3/envs/mne/lib/python3.8/site-packages/mne/source_estimate.py", line 651, in plot
    brain = plot_source_estimates(
  File "<decorator-gen-153>", line 24, in plot_source_estimates
  File "/home/adam2392/miniconda3/envs/mne/lib/python3.8/site-packages/mne/viz/_3d.py", line 1765, in plot_source_estimates
    backend = _get_3d_backend()
  File "/home/adam2392/miniconda3/envs/mne/lib/python3.8/site-packages/mne/viz/backends/renderer.py", line 153, in _get_3d_backend
    raise RuntimeError(f'Could not load any valid 3D backend: '
RuntimeError: Could not load any valid 3D backend: pyvista, mayavi, notebook
```

mne sys_info:
```
mne sys_info
Platform:      Linux-5.5.4-050504-generic-x86_64-with-glibc2.17
Python:        3.8.10 (default, Jun  4 2021, 15:09:15)  [GCC 7.5.0]
Executable:    /home/adam2392/miniconda3/envs/mne/bin/python
CPU:           x86_64: 24 cores
Memory:        62.7 GB

mne:           0.23.0
numpy:         1.20.3 {blas=openblas, lapack=openblas}
scipy:         1.6.3
matplotlib:    3.3.2 {backend=Qt5Agg}

sklearn:       0.24.2
numba:         Not found
nibabel:       3.2.1
nilearn:       0.8.0
dipy:          Not found
cupy:          Not found
pandas:        1.2.4
mayavi:        Not found
pyvista:       0.31.1 {OpenGL 4.5.0 NVIDIA 460.80 via GeForce GTX 750/PCIe/SSE2}
vtk:           9.0.1
PyQt5:         5.15.4
```

hoechenberger, 2021-06-17:
How did you create the MNE environment? As described in the docs?

hoechenberger, 2021-06-17:
@GuillaumeFavelier might be able to help here

hoechenberger, 2021-06-17:
A few months ago, it appears @dengemann was using Rosetta to run PyVista on his M1:
https://twitter.com/dngman/status/1343901324908359684

adam2392, 2021-06-17:
> How did you create the MNE environment? As described in the docs?

On my **Linux**, yes using conda environment. I tried again with Python3.6, and it worked... so not sure what's up. Maybe python3.8 doesn't play nicely w/ pyvista/gui stuff?

I should specify, I'm porting over the connectivity into `mne-connectivity`, but for all intents and purposes the environments/etc. are all the same. However, now when I run things it gets hung on `mne_inverse_envelope_correlation.py`. I'm a 100% noob at sphinx, so not sure if there's a way to increase visibility into the background processes happening to debug if this is a hung process, etc.?

```
(mne3.6) adam2392@adam2392-system-hackerman:~/Documents/mne-connectivity$ make build-doc
cd doc; make clean
make[1]: Entering directory '/home/adam2392/Documents/mne-connectivity/doc'
rm -rf _build/*
rm -rf auto_examples/
rm -rf generated/*
rm -rf modules/generated/*
make[1]: Leaving directory '/home/adam2392/Documents/mne-connectivity/doc'
cd doc; make html
make[1]: Entering directory '/home/adam2392/Documents/mne-connectivity/doc'
sphinx-build -b html -d _build/doctrees  -nWT --keep-going . _build/html
Running Sphinx v4.0.2
making output directory... done
checking for /home/adam2392/Documents/mne-connectivity/doc/references.bib in bibtex cache... not found
parsing bibtex file /home/adam2392/Documents/mne-connectivity/doc/references.bib... parsed 9 entries
[autosummary] generating autosummary for: api.rst, bibliography.rst, index.rst, install.rst, whats_new.rst
[autosummary] generating autosummary for: /home/adam2392/Documents/mne-connectivity/doc/generated/mne_connectivity.envelope_correlation.rst, /home/adam2392/Documents/mne-connectivity/doc/generated/mne_connectivity.phase_slope_index.rst, /home/adam2392/Documents/mne-connectivity/doc/generated/mne_connectivity.spectral_connectivity.rst
loading intersphinx inventory from https://docs.python.org/3/objects.inv...
loading intersphinx inventory from https://mne.tools/dev/objects.inv...
loading intersphinx inventory from https://mne.tools/mne-bids/dev/objects.inv...
loading intersphinx inventory from https://numpy.org/devdocs/objects.inv...
loading intersphinx inventory from https://scipy.github.io/devdocs/objects.inv...
loading intersphinx inventory from https://matplotlib.org/objects.inv...
loading intersphinx inventory from http://pandas.pydata.org/pandas-docs/dev/objects.inv...
loading intersphinx inventory from http://scikit-learn.org/stable/objects.inv...
loading intersphinx inventory from https://docs.pyvista.org/objects.inv...
loading intersphinx inventory from https://joblib.readthedocs.io/en/latest/objects.inv...
loading intersphinx inventory from https://nipy.org/nibabel/objects.inv...
loading intersphinx inventory from http://nilearn.github.io/objects.inv...
intersphinx inventory has moved: http://scikit-learn.org/stable/objects.inv -> https://scikit-learn.org/stable/objects.inv
intersphinx inventory has moved: http://pandas.pydata.org/pandas-docs/dev/objects.inv -> https://pandas.pydata.org/pandas-docs/dev/objects.inv
generating gallery...
Using Sphinx-Gallery to convert rst text blocks to markdown for .ipynb files.
generating gallery for auto_examples... [ 44%] mne_inverse_envelope_correlation.py  
```
^ been there for like an hour.

GuillaumeFavelier, 2021-06-22:
Does it work when you run the example? Does the plot hang only during doc building? :thinking: