mne-tools/mne-python #9475: How to handle untested file versions with read raw body?

rob-luke, 2021-06-14:
On the discourse forum a user has tried to load data with the read_raw_boxy function using a version of data that we haven't tested.

https://mne.discourse.group/t/error-using-mne-for-fnirs-boxy-data/3247

Currently this throws an error. Should we change this to a warning? And ask users to report to us if it's working correctly and maybe contribute test data?

I'm happy to make the change if that's what we should do.

larsoner, 2021-06-15:
We have a mix of both warn-if-unknown and error-if-unknown. If we think it's more likely to work okay rather than fail silently and give an incorrect result, I'm okay with a warning instead

rob-luke, 2021-06-15:
I think its most likely to fail or give an incorrect result. There is already some wrangling in the code to handle changes in the file format. For example...

https://github.com/mne-tools/mne-python/blob/25f60579aca14e879a98399d1c083c70f9ae1327/mne/io/boxy/boxy.py#L110-L113

I guess we wait and see what Parveen comes back with,