mne-tools/mne-python #9479: find_bad_channels_maxwell breaks if during some chunk all channels are flat

levchenkoegor, 2021-06-15:
Hello everyone!

I faced with a problem using `mne.preprocessing.find_bad_channels_maxwell()`. At some point it returns an error:
```
        Interval 103:  510.000 -  514.999
            Flat ( 2): MEG1243 MEG1532
        Interval 104:  515.000 -  519.999
Traceback (most recent call last):
  File "<input>", line 172, in <module>
  File "<decorator-gen-415>", line 24, in find_bad_channels_maxwell
  File "E:\Egor_Levchenko\Software\Miniconda\envs\MCCA-MEG\lib\site-packages\mne\preprocessing\maxwell.py", line 2190, in find_bad_channels_maxwell
    delta = chunk_raw.get_data(these_picks)
  File "<decorator-gen-170>", line 22, in get_data
  File "E:\Egor_Levchenko\Software\Miniconda\envs\MCCA-MEG\lib\site-packages\mne\io\base.py", line 819, in get_data
    picks = _picks_to_idx(self.info, picks, 'all', exclude=())
  File "E:\Egor_Levchenko\Software\Miniconda\envs\MCCA-MEG\lib\site-packages\mne\io\pick.py", line 1058, in _picks_to_idx
    '(%r)' % (orig_picks,))
ValueError: No appropriate channels found for the given picks ([])
```

I checked the code and my data. I think the problem is that my data is corrupted a little bit and I have all flat channels for around  20 seconds. After detecting the flat channels the code takes only `good_meg_picks` which are not in `chunk_flats`. As a result, `these_picks` appears to be empty if all channels are in `chunk_flats`:

https://github.com/mne-tools/mne-python/blob/8b9616dfbc377ce468a6d52b43f7a973eac47cab/mne/preprocessing/maxwell.py#L2167-L2175
https://github.com/mne-tools/mne-python/blob/8b9616dfbc377ce468a6d52b43f7a973eac47cab/mne/preprocessing/maxwell.py#L2187

You can see a snapshot of raw data below:

[image]

I think this is more a problem of my data, but I guess it should be nice to ignore "all flat" chunks with a warning or something like that. Correct me please if I am wrong and thank you for any comments!

System properties:
```
OS: Windows Server 2012 R2
python=3.7.7
mne=0.22.0
```

agramfort, 2021-06-16:
can you try
https://mne.tools/stable/generated/mne.preprocessing.annotate_flat.html
before running maxfilter?
it should make maxfilter ignore the flat segment

levchenkoegor, 2021-06-16:
Thank you! 
I tried running it, but it is strange that flat segments are not fully detected:
```
>>> annot, bads = mne.preprocessing.annotate_flat(raw, picks='meg')

Finding flat segments
100%|██████████| Channels : 306/306 [00:04<00:00,   62.14it/s]
Marking 0.44% of time points (371 segments) and 0/306 channels bad
```

It detects only specific pieces in one part of the big flat segment:
[image]

I've uploaded the data sample [here](https://drive.google.com/drive/folders/1QTtMJIeBDsi5JlauglL0ZkkOOQeSbOn3?usp=sharing).

larsoner, 2021-06-16:
Looks like there are likely bugs with `annotate_flat` *and* `find_bad_channels_maxwell`, I'll look into it!

larsoner, 2021-07-06:
> > mne=0.22.0

> I tried running it, but it is strange that flat segments are not fully detected:

I cannot reproduce on 0.22.0:
```
import mne
raw = mne.io.read_raw_fif('sub-1011_ses-20190430_task-vid2_meg.fif').load_data()
annot, bads = mne.preprocessing.annotate_flat(raw, picks='meg', bad_percent=100, verbose=True)
raw.set_annotations(annot)
raw.plot()
```
outputs:
```
Marking 8.85% of time points (5 segments) and 0/306 channels bad
```
Either way, though, I recommend updating to the latest release 0.23 as 0.22 is no longer supported. There I get the same thing:

[image]

Can you try on 0.23? If you have the same problem with the code snippet above I'll boot into Windows to see if I can replicate there.

closed 2021-07-06 by agramfort

levchenkoegor, 2021-07-08:
@larsoner, thank you! 

I have updated mne to 0.23.0 and tried one more time.

I think the main issue was that I hadn't specified the `bad_percent` parameter, with `bad_percent=100` the detection for that file works well:
```
import mne

meg_file_subj = 'sub-1011_ses-20190430_task-vid2_meg.fif'

raw = mne.io.read_raw_fif(meg_file_subj, preload=True)
annot, _ = mne.preprocessing.annotate_flat(raw, picks='meg', bad_percent=100, verbose=True)
raw.set_annotations(annot)
raw.plot()
```
[image]

At the same time, I face with one more issue. I ran exactly the same code for one more subject and it marks the whole file as bad:

```
import mne

meg_file_subj = 'sub-1003_ses-20200812_task-vid1_meg.fif'
...

Finding flat segments
100%|██████████| Channels : 306/306 [00:07<00:00,   40.27it/s]
Marking 100.00% of time points (2 segments) and 0/306 channels bad
```
[image]

I filtered the data with `raw.filter(1, 40)` to be sure that it looks fine and that's what I got:
```
import mne

meg_file_subj ='sub-1003_ses-20200812_task-vid1_meg.fif'

raw = mne.io.read_raw_fif(meg_file_subj, preload=True)
raw.filter(1, 40)
annot, _ = mne.preprocessing.annotate_flat(raw, picks='meg', bad_percent=100, verbose=True)
raw.set_annotations(annot)
raw.plot()

Finding flat segments
100%|██████████| Channels : 306/306 [00:06<00:00,   45.23it/s]
Marking 2.09% of time points (2159 segments) and 0/306 channels bad
```

[image]

I am not sure that this is the expected behaviour of the function, so I decided to comment.

Here the whole conda-environemnt.yml file that I use (maybe it's useful):
```
name: MCCA-MEG
dependencies:
  **- python=3.7.7**
  - pip=20.2.4
  - numpy=1.17.2
  - scipy=1.3.1
  - matplotlib=3.3.2
  - pandas=1.1.1
  - scikit-learn=0.24.0
  - seaborn=0.11.1
  **- conda-forge::mne=0.23.0**
  - conda-forge::mne-bids=0.6
  - conda-forge::codebraid=0.5.0
  - conda-forge::pandoc=2.14
  - conda-forge::ipykernel=5.3.4
  - pip:
    - pybids==0.12.3
```

larsoner, 2021-07-08:
You probably want `min_duration` set to something higher. Our default (0.001) is probably too small...

PR #9531 (2021-07-06): MRG, BUG: Deal with flats better in maxwell autobad
Just add a test that things behave how we expect for #9479, and a warning to help people. We could maybe do something smarter than just quit, but I think it makes sense to just point people to `annotate_flat` so that they can fix things there instead. With this code a variant of the minimal example from #9479 works as expected:
```
import mne
raw = mne.io.read_raw_fif('sub-1011_ses-20190430_task-vid2_meg.fif').load_data()
raw.set_channel_types(dict(EEG061='eog', EEG062='eog'))
annot, _ = mne.preprocessing.annotate_flat(raw, bad_percent=100, v [... cut]