mne-tools/mne-python #9482: BUG: weird CNT reading

kingjr, 2021-06-16:
Tried on both stable and dev.

I'm trying to read a [.cnt  file](https://osf.io/xq5pd/). 

MNE indicates no error or warnings.

However, the raw (or epoched data, based on the annotations) looks very weird: 

* every other channels alternate between a small set of discrete values: ```list(map(lambda x: len(np.unique(x)), raw._data[:, :1000]))``` => ```[[994, 8, 992, 7, 996, 6, 996, 7, 992, 6 ...]]```
* the data within a given raw channels is discontinuous and does not show the classic 1/f dynamics:
[image]
* the epoched data based on the annotation does not look any kind of ERP: 
[image]

I suspect there is something wrong in how the data chunks are read and assembled, but I don't really understand the mne.io.cnt parser to help :/

larsoner, 2021-06-16:
Indeed it seems like a likely datatype or order problem, I'll take a look in the coming days

larsoner, 2021-06-28:
CNT does not reliably store how many bytes per sample, and we do warn during runtime:
```
/Users/larsoner/opt/miniconda3/bin/mne:33: RuntimeWarning: Could not define the number of bytes automatically. Defaulting to 2.
```
Using `data_format='int32'` things look okay:
```
$ python -ic "import mne; raw = mne.io.read_raw_cnt('STIR1.cnt', data_format='int32'); raw.plot()"
```
[image]

Closing as I'm not sure any fix is needed at the MNE end. Maybe a doc update with a warning (?), but to me the runtime warning was clear enough to work around the format limitation.

closed 2021-06-28 by larsoner

kingjr, 2021-07-06:
awesome, thanks @larsoner !