mne-tools/mne-python #9483: Trying to apply rejection thresholds repeatedly to Epochs crashes if only using a subset of channels for a subsequent reject

hoechenberger, 2021-06-16:
MWE:
```python
from pathlib import Path
import mne

sample_dir = Path(mne.datasets.sample.data_path())
sample_fname = sample_dir / 'MEG' / 'sample' / 'sample_audvis_raw.fif'

raw = mne.io.read_raw_fif(sample_fname)
raw.crop(tmax=60)

events = mne.find_events(raw, stim_channel='STI 014')
event_id = {'auditory/left': 1, 'auditory/right': 2, 'visual/left': 3,
            'visual/right': 4, 'face': 5, 'buttonpress': 32}

epochs = mne.Epochs(raw, events=events, event_id=event_id,
                    tmin=-0.2, tmax=0.5, baseline=(None, 0),
                    preload=True)

# This works as expected
reject_a = dict(grad=8e-10, mag=10e-11)
epochs.drop_bad(reject=reject_a)

# This raises an exception:
# ValueError: reject["mag"] == inf > 1e-10 (old value), new
# reject values must be at least as stringent as previous ones
reject_b = dict(grad=1e-10)
epochs.drop_bad(reject=reject_b)

# This works
reject_c = dict(grad=1e-10, mag=9e-11)
epochs.drop_bad(reject=reject_c)
```

drammock, 2021-06-16:
So just to be clear, the bug is that `reject_c` works when it shouldn't.  Right?

hoechenberger, 2021-06-16:
Sorry, obviously I wasn't clear enough.

The issue is that `reject_b` doesn't work when it shouldn't; `reject_c` works as expected.

The issue is that when calling `drop_bad()` with `reject_a`, we pass a dict that contains PTP thresholds for mag and grad. If I subsequently want to do another round of PTP rejection, but only for one of the initially used channel types, `drop_bad()` raises an error message, claiming that the PTP threshold for channel type I specified earlier (but didn't specify now) is `inf`.

drammock, 2021-06-16:
sorry, I was mixed up about thresholds being "more extreme" versus "more stringent".  Yes, seems like a bug in handling of `reject_b`.

agramfort, 2021-06-16:
PR welcome

>

PR #9485 (2021-06-17): MRG, FIX: Allow multiple runs of Epochs.drop_bad() even if only using a subset of initial reject channels
Closes #9483
commit 22b95895 (2021-06-17): MRG, FIX: Allow multiple runs of Epochs.drop_bad() even if only using a subset of initial reject channels (#9485)

* Allow multiple runs of Epochs.drop_bad() even if only using a subset of initial reject channels

Fixes #9483

* Update doc/changes/latest.inc [skip azp][skip github]

* Refactor
closed 2021-06-17 by drammock