mne-tools/mne-python #9488: Adding neural networks as inverse solutions

LukeTheHecker, 2021-06-18:
#### Describe the new feature or enhancement
**The feature**
Currently, there are a handful of M/EEG source localization methods available within mne-python: minimum norm estimates and derivatives (dSPM, eLORETA...), equivalent current dipole, sparse bayesian approaches and beamforming. To this list I would like to add artificial neural networks (ANNs) as demonstrated in Hecker et al., 2021. A working implementation outside of MNE can be found [here](https://github.com/lukethehecker/esinet). 

**Why do I think it's useful?**
This feature would add a new approach to the inverse problem, which has shown to work well on simulated data. An implementation within MNE would hopefully allow users to test and compare the ANN-based inverse solutions with the existing and more established approaches using their own data.

#### Additional comments
I would like to discuss this feature with the mne-python community. Let's say I'll implement this feature such that the devs are satisfied with it. How will maintainence work in the long run? For example, if mne-python receives  major updates in the upcoming years which cause my code to not work anymore - how do you proceed in these cases? My concern is that submodules like this one may clutter the mne-python repository if I don't personally commit to maintainance for an elongated period. I'm happy to hear your thoughts about it!

Literature:
Hecker, L., Rupprecht, R., Tebartz van Elst, L., & Kornmeier, J. (2021). ConvDip: A convolutional neural network for better EEG Source Imaging. Frontiers in Neuroscience, 15, 533.

agramfort, 2021-06-18:
@LukeTheHecker I would really advise to maintain this in a separate package that is fully compatible with mne.

basically you would have

```
from my_package import my_great_inverse_method

stc = my_great_inverse_method(fwd, evoked, ...)

stc.plot()
```

the main reason is the dependency on deep learning toolkits in our CIs that would really like to limit due to an already super long test time.

LukeTheHecker, 2021-06-18:
That makes sense to me, thank you for your fast reply!

closed 2021-06-18 by LukeTheHecker