mne-tools/mne-python #9495: Plotting 20 ICA topomaps, makes the ica name appear on the topomap above in small screens

pooyaEst, 2021-06-22:
#### Describe the bug
Plotting 20 ICA topomaps, makes the ica name appear on the topomap above in small screens and clicking on the name to remove it, also opens the page related to topomap above.

[image]

#### Expected results
 being able to either reduce size of topomaps or number of them appearing in a single page or both.

cbrnr, 2021-06-22:
What happens if you resize the window, e.g. make it really small and then enlarge it again? Does it change anything?

pooyaEst, 2021-06-23:
if i change the size to around half in my 13" display, it works but still that would be nice to have an option to modify number of components present or their size.

cbrnr, 2021-06-23:
You can set the number of components with the `picks` argument (https://mne.tools/stable/generated/mne.preprocessing.ICA.html#mne.preprocessing.ICA.plot_components).

closed 2024-07-27 by sappelhoff