mne-tools/mne-python #9496: Describe effect of channel type in more detail

robintibor, 2021-06-22:
#### Proposed documentation enhancement

It would be nice to prominently signal that setting of channel type influences whether certain functions are applied like filter and also have a clear documentation what the exact effects of the channel types are. At least we understood from a chat with @agramfort today, that, e.g., setting 'misc' channel type will mean highpass/lowpass will not be applied. Before the chat I was not aware of that, and also now I could not find a detailed documentation about precise effects of setting different channel types. So could be supernice to have some further documentation on the precise effects and maybe link to this in some prominent places?

agramfort, 2021-06-22:
this picks docstring in Raw.filter is quite explicit I think but I agree you need to know where to look

https://mne.tools/stable/generated/mne.io.Raw.html#mne.io.Raw.filter

robintibor, 2021-06-22:
and this default behavior will only happen for filter or also for other functions?

agramfort, 2021-06-22:
what function do you have in mind? the docstring should be explicit
otherwise we need to fix it.

apply_function also is applied to only data channels.

>

robintibor, 2021-06-22:
I did not have a specific function in mind. Rather had in mind, if we go with creating "target" channels in Braindecode with misc data type if any preprocessing the user may apply may cause surprises or not. Seems like in the current example our guys try to be explicit:
https://github.com/braindecode/braindecode/pull/261/files#diff-8b00a328d1fc4cdfbde54aaf6ef761f3d810e51983dd18cc687bc3cce7b9478bR56-R63
But who knows what users will do :) :)

agramfort, 2021-06-23:
let me know how I can help

>

robintibor, 2021-06-23:
Are you aware of any functions in mne that if applied to the raw by default will also transform 'misc' channels and where that transformation is likely to be undesired if the 'misc' channels contain targets? Otherwise also feel fine to close this issue or leave it open if you think the channel type effects should be more prominently highlighted somewhere.

agramfort, 2021-06-23:
it's too hard to answer. If you resample, you touch all channels. If you filter it's different. It depends on the need more than the software spec

robintibor, 2021-06-23:
oki I see thanks. as said feel free to close issue or leave open as you wish.

closed 2021-06-23 by agramfort