mne-tools/mne-python #9502: plotting overlapping epochs: Events not properly aligned in plot?

sappelhoff, 2021-06-24:
I have to create overlapping epochs from scratch using [EpochsArray](https://mne.tools/dev/generated/mne.EpochsArray.html#mne.EpochsArray). I designed my events, tmin, and other parameters so that the main event for each epoch (the t0 event) is exactly in the middle of a 2 second timespan (-1s to 1s). However when plotting via `epochs.plot`, the events are **not** in the middle.

This plotting appearance is fixed when I subset my epochs to only contain those epochs that are not overlapping.

The following MWE is modified from [the docs](https://mne.tools/dev/auto_tutorials/simulation/10_array_objs.html#creating-mne-python-data-structures-from-scratch).

The only change from the docs example is to go from non-overlapping epochs: `np.arange(0, 1000, sampling_freq)` 
to overlapping epochs: `np.linspace(0, 200, 5, dtype=int)`.

```python
import mne
import numpy as np

sampling_freq = 200  # in Hertz

info = mne.create_info(ch_names=['10 Hz sine', '5 Hz cosine'],
                       ch_types=['misc'] * 2,
                       sfreq=sampling_freq)

times = np.linspace(0, 1, sampling_freq, endpoint=False)
sine = np.sin(20 * np.pi * times)
cosine = np.cos(10 * np.pi * times)

data = np.array([[0.2 * sine, 1.0 * cosine],
                 [0.4 * sine, 0.8 * cosine],
                 [0.6 * sine, 0.6 * cosine],
                 [0.8 * sine, 0.4 * cosine],
                 [1.0 * sine, 0.2 * cosine]])

events = np.column_stack((np.linspace(0, 200, 5, dtype=int),  # NOTE: used to be np.arange(0, 1000, sampling_freq),
                          np.zeros(5, dtype=int),
                          np.array([1, 2, 1, 2, 1])))
event_dict = dict(condition_A=1, condition_B=2)
simulated_epochs = mne.EpochsArray(data, info, tmin=-0.5, events=events,
                                   event_id=event_dict)

simulated_epochs.plot(picks='misc', events=events, event_id=event_dict)

```

Take a look at the resulting plot. It's hard to see in the following screenshot, but the events are NOT centered in the middle of the epoch -> rather, they are clustered at the end of the plot:

[image]

When subsetting these epochs to only contain non-overlapping epochs, the issue is solved and the plot shows the expected alignment of events in the middle of epochs.

```python
simulated_epochs[np.array((0, 4))].plot(picks='misc', events=events[np.array((0, 4))], event_id=event_dict)

```

[image]

**Is this somehow expected? Or is it a bug?**

drammock, 2021-06-24:
there is definitely a bug, but also a difficult question: if a single event would be visible in multiple epochs, should it be shown in all of them?  Options:

1. yes, show an event in multiple epochs if appropriate
2. no, only show it in the sequentially first epoch where it appears
3. no, only show it in the epoch that it "belongs" to
 
I think number 3 is impossible because `events` are passed in separately so we would have to guess/infer which event "belongs" to which epoch (not so easy for things like errant button presses, or in cases like this where epochs overlap).  I will try to solve this as if number 1 is the correct approach, but please weigh in if you think number 2 is actually better.  Another question is whether we should log an info or warning message when events are duplicated in the figure?

sappelhoff, 2021-06-24:
Thanks for picking this up! I think your option 1 is most appropriate, and I agree that a log info is probably a good idea as well.

PR #9505 (2021-06-25): Fix plotting of events when there are overlapping epochs
closes #9502 

cc @sappelhoff
closed 2021-06-25 by agramfort