mne-tools/mne-python #9507: Is it possible to run test suite against installed mne?

bloyl, 2021-06-26:
As part of [mne-tools/mne-docker](https://github.com/mne-tools/mne-docker), I'd like to run the `mne` test suite against the installed version of `mne`. This is complicated by the fact that our tests are nested within the mne application code. See https://docs.pytest.org/en/6.2.x/goodpractices.html

I've tried launching from outside of the source tree (i.e. `pytest mne-python-devel/mne/tests/test_evoked.py::test_io_evoked --verbose`) but pytest still seems to pick up the devel version in the `mne-python-devel` directory as opposed to  the installed version. 

Has anyone tried this or have any good ideas on how to trick it into working?

agramfort, 2021-06-26:
we use full imports (not relative) in tests so I would expect it to work.
But I have not tested...

>

bloyl, 2021-06-27:
Unfortunately `pytest` does some magic with the `sys.path` so it can import tests etc (https://docs.pytest.org/en/6.2.x/pythonpath.html).

Using `--import-mode prepend` is the default and results in running `mne` out of the source tree.  `--import-mode append` also fails. It think fails because of conflicts between the installed `conftest.py`/tests and the copies in the source tree.

Anyway its not a big deal, but if anyone has any ideas let me know.

Thanks.

larsoner, 2021-06-28:
I added a `raise RuntimeError('fail')` to `mne/__init__.py`, and then `pytest python/mne-python/mne/tests/test_morph.py -x` fails as expected but `--import-mode importlib` works. You need pytest 6 for this. Did you try it?

bloyl, 2021-06-28:
`--import-mode importlib` does indeed work and for some unknown reason I had not tried it 🤦. It was right there on the doc page I linked to.

Thanks @larsoner

closed 2021-06-28 by bloyl