mne-tools/mne-python #9510: Add fiducial point check in _read_xyz

sans-dev, 2021-06-28:
Hi everyone. 
Would it be a good idea to add a check for fiducial points in `_read_xyz` function? Right now I am dealing with a custom montage in .csv format where nasion, lpa and rpa coordiantes are written into the csv. By checking for some standard namings ('nasion', 'nz', 'lpa', 'rpa') in` ch_pos` we could extract their coordinates and pass them into `make_dig_montage`.

https://github.com/mne-tools/mne-python/blob/3078533f8704820a168797578f5b47ddff4be347/mne/channels/_standard_montage_utils.py#L345

agramfort, 2021-06-28:
are you relying on a private function for your script?

>

sans-dev, 2021-06-28:
No, I am calling `read_custom_montage`, which then internally uses the `_read_xyz` function based on the file extension (csv in this case).

agramfort, 2021-06-29:
we could expose some extra parameters in read_custom_montage for this.

I am reluctant to make too much magic

>

sans-dev, 2021-06-30:
Parametrization is a good idea. I suggest a list of string labels. 
```
_read_theta_phi_in_degrees(fname, head_size, fid_names=None,
                               add_fiducials=False):
```
_read_theta_phi_in_degrees already provides such an interface. We could transfer it to _read_xyz. The initialization would happen when read_custom_montage is called. We could use default arguments there.

agramfort, 2021-06-30:
I would add new parameters to read_custom_montage

what we do in functions that start with _ is not a problem. It's private so
we
can do whatever we want there. What matters is the changes to
read_custom_montage