mne-tools/mne-python #9511: evoked.plot_topomap extrapolation can fail when few electrodes present

drammock, 2021-06-28:
from the forum: https://mne.discourse.group/t/erp-plot-topomap-nan-to-integer-error/3317

extrapolate = 'box' or 'local' both work, but the default ('head') fails (at some point in the code we call `np.arcsin` and its argument is out of the domain for that function, thus introducing a NaN)

```py
import numpy as np
import mne

exp3 = mne.io.read_raw_edf('exp3.edf', preload=True)
exp3.filter(1,30, method='iir')
events_exp3 = mne.make_fixed_length_events(exp3, duration=1.0)

ten_five = mne.channels.make_standard_montage('standard_1005')
exp3.set_montage(ten_five)

epochs_exp3 = mne.Epochs(
    exp3, events_exp3, event_id={'resting_state': 1}, tmin=-0.3, tmax=0.7,
    preload=True)

rest_exp3 = epochs_exp3['resting_state'].average()
fig = rest_exp3.plot_topomap()
```
Traceback:
```
---------------------------------------------------------------------------
ValueError                                Traceback (most recent call last)
<ipython-input-4-7729af823da6> in <module>
----> 1 fig = rest_exp3.plot_topomap()

/opt/mne/python/mne/evoked.py in plot_topomap(self, times, ch_type, vmin, vmax, cmap, sensors, colorbar, scalings, units, res, size, cbar_fmt, time_unit, time_format, proj, show, show_names, title, mask, mask_params, outlines, contours, image_interp, average, axes, extrapolate, sphere, border, nrows, ncols)
    443                      axes=None, extrapolate=_EXTRAPOLATE_DEFAULT, sphere=None,
    444                      border=_BORDER_DEFAULT, nrows=1, ncols='auto'):
--> 445         return plot_evoked_topomap(
    446             self, times=times, ch_type=ch_type, vmin=vmin,
    447             vmax=vmax, cmap=cmap, sensors=sensors, colorbar=colorbar,

/opt/mne/python/mne/viz/topomap.py in plot_evoked_topomap(***failed resolving arguments***)
   1714         adjust_for_cbar = colorbar and ncols is not None and idx >= ncols - 1
   1715         ax_idx = idx + 1 if adjust_for_cbar else idx
-> 1716         tp, cn, interp = _plot_topomap(
   1717             data[:, idx], pos, axes=axes[ax_idx],
   1718             mask=mask_[:, idx] if mask is not None else None, **kwargs)

/opt/mne/python/mne/viz/topomap.py in _plot_topomap(data, pos, vmin, vmax, cmap, sensors, res, axes, names, show_names, mask, mask_params, outlines, contours, image_interp, show, onselect, extrapolate, sphere, border, ch_type, cnorm)
    950 
    951     # find mask limits
--> 952     extent, Xi, Yi, interp = _setup_interp(
    953         pos, res, extrapolate, sphere, outlines, border)
    954     interp.set_values(data)

/opt/mne/python/mne/viz/topomap.py in _setup_interp(pos, res, extrapolate, sphere, outlines, border)
    837     yi = np.linspace(ymin, ymax, res)
    838     Xi, Yi = np.meshgrid(xi, yi)
--> 839     interp = _GridData(pos, extrapolate, clip_origin, clip_radius, border)
    840     extent = (xmin, xmax, ymin, ymax)
    841     return extent, Xi, Yi, interp

/opt/mne/python/mne/viz/topomap.py in __init__(self, pos, extrapolate, origin, radii, border)
    607 
    608         # Adding points outside the extremes helps the interpolators
--> 609         outer_pts, mask_pts, tri = _get_extra_points(
    610             pos, extrapolate, origin, radii)
    611         self.n_extra = outer_pts.shape[0]

/opt/mne/python/mne/viz/topomap.py in _get_extra_points(pos, extrapolate, origin, radii)
    576         # return points on the head circle
    577         angle = np.arcsin(distance / np.mean(radii))
--> 578         n_pnts = max(12, int(np.round(2 * np.pi / angle)))
    579         points_l = np.linspace(0, 2 * np.pi, n_pnts, endpoint=False)
    580         use_radii = radii * 1.1 + distance

ValueError: cannot convert float NaN to integer
> /opt/mne/python/mne/viz/topomap.py(578)_get_extra_points()
    576         # return points on the head circle
    577         angle = np.arcsin(distance / np.mean(radii))
--> 578         n_pnts = max(12, int(np.round(2 * np.pi / angle)))
    579         points_l = np.linspace(0, 2 * np.pi, n_pnts, endpoint=False)
    580         use_radii = radii * 1.1 + distance
```
cc @mmagnuski

mmagnuski, 2021-07-06:
Fixed by #9528 but I didn't link this issue correctly for the autoclose.
BTW @drammock @larsoner - is there a neat way of linking multiple issues to be autoclosed when the PR is merged (without writing `fixes #xxxx, fixes #xxxx, fixes #xxxx`)?

closed 2021-07-06 by mmagnuski

larsoner, 2021-07-06:
I always just write each "Closes" on a separate line. I've never had more than maybe 5 and even then having the newlines is satisfying, so I haven't looked too hard for a more compact solution :)

mmagnuski, 2021-07-06:
@larsoner Yes, I was just curious :) I checked and there is no way around `fixes #xxxx, fixes #xxxx, fixes #xxxx` ([stack overflow link](https://stackoverflow.com/questions/3547445/closing-multiple-issues-in-github-with-a-commit-message)).