mne-tools/mne-python #9523: Add a method to rename annotation descriptions

rob-luke, 2021-07-05:
#### Describe the new feature or enhancement

I find myself repeatedly needing to explain how to rename annotations to end users. I rename annotations using something like...

```python
ann = raw_intensity.annotations
ann.description = [d.replace('1.0', 'Control') for d in ann.description]
ann.description = [d.replace('2.0', 'Tapping/Left') for d in ann.description]
ann.description = [d.replace('3.0', 'Tapping/Right') for d in ann.description]
raw_intensity.set_annotations(ann)
```

Which isn't much code. But the barrier to entry is quite high. It requires that a user understands list comprehensions, which I think is an advanced topic. New users testing out MNE usually first want to load their data, rename triggers, and plot the data. So I suggest we add a method to annotations for renaming descriptions.

Thoughts?

#### Describe your proposed implementation
I would add an `annotations.rename(old_name, new_name)` method.

#### Describe possible alternatives
Leave as it is

#### Additional comments
I am happy to write this if others think its worthwhile

cbrnr, 2021-07-05:
Good idea!

drammock, 2021-07-05:
Would be nice if you could update multiple at once, sort of like dict.update or raw.rename_channels

rob-luke, 2021-07-05:
> Would be nice if you could update multiple at once, sort of like dict.update or raw.rename_channels

I thought the same thing and had suggested that as an alternative in the PR.

PR #9525 (2021-07-08): MRG: Add annotations.rename()
#### Reference issue
Closes #9523

#### What does this implement/fix?
Adds a `annotations.rename()` method and `raw.rename_annotations()`.

#### Additional information
~Currently the code takes the old name and the new name as arguments. An alternative suggestion would be to passing in a mapping like `{"old_name": "new_name", "another_old_name": "fancy_new_name"}`. I’m on the fence about which way to implement it, happy to change if you want.~
closed 2021-07-08 by cbrnr