mne-tools/mne-python #9527: Annotation and crop - different onset points after reloading the saved croppend data

kcsaba12, 2021-07-05:
#### Describe the bug
I suspect that this issue is similar to the #9383.
The issue only courses when a raw instance created with Annotation. Originally labeled data as brain vision files are work fine.
Cropping the instance, saving and reloading it shows the annotations in the original place before the cropping.

#### Steps to reproduce
```
import numpy as np
import mne
CH_NUM = 32
FS = 160

data = np.zeros((CH_NUM, FS * 20))
ch_names = [f'EEG{i}' for i in range(CH_NUM)]
ch_types = ['eeg'] * len(ch_names)
onset = [8, 12, 18]
duration = [2] * len(onset)
description = list(range(len(onset)))

info = mne.create_info(ch_names, ch_types=ch_types, sfreq=FS)
raw = mne.io.RawArray(data, info)
annotation = mne.Annotations(onset, duration, description)
raw = raw.set_annotations(annotation)

sess = raw.copy()
sess.crop(5, 18)
file = 'test_file_raw.fif'
sess.save(file, overwrite=True)
sess.plot(block=False)
r = mne.io.read_raw(str(file))
r.plot(block=False)
raw.plot(block=True)
```

#### Expected results
I expect that the annotated data should look like the same before and after saving it.

#### Actual results
The annotations are shifted...

#### Additional information
Platform:      Windows-10-10.0.19041-SP0
Python:        3.8.5 (default, Sep  3 2020, 21:29:08) [MSC v.1916 64 bit (AMD64)]
Executable:    C:\Programs\Miniconda3\envs\bci2\python.exe
CPU:           Intel64 Family 6 Model 158 Stepping 10, GenuineIntel: 12 cores
Memory:        Unavailable (requires "psutil" package)
mne:           0.23.0
numpy:         1.19.3 {blas=D:\\a\\1\\s\\numpy\\build\\openblas_info, lapack=D:\\a\\1\\s\\numpy\\build\\openblas_lapack_info}
scipy:         1.6.0
matplotlib:    3.3.3 {backend=TkAgg}
sklearn:       0.24.0
numba:         Not found
nibabel:       Not found
nilearn:       Not found
dipy:          Not found
cupy:          Not found
pandas:        1.2.0
mayavi:        Not found
pyvista:       Not found
vtk:           Not found

kcsaba12, 2021-07-08:
[image]

agramfort, 2021-07-08:
I confirm the bug... you can fix it if you set a meas_date with the raw eg with:

```python
import numpy as np
import mne
from datetime import datetime, timezone

CH_NUM = 32
FS = 160

data = np.zeros((CH_NUM, FS * 20))
ch_names = [f'EEG{i}' for i in range(CH_NUM)]
ch_types = ['eeg'] * len(ch_names)
onset = [8, 12, 18]
duration = [2] * len(onset)
description = list(range(len(onset)))

info = mne.create_info(ch_names, ch_types=ch_types, sfreq=FS)
raw = mne.io.RawArray(data, info)
annotation = mne.Annotations(onset, duration, description)
raw = raw.set_annotations(annotation)

raw.set_meas_date(datetime.now(tz=timezone.utc))

sess = raw.copy()
sess.crop(5, 18)
file = 'test_file_raw.fif'
sess.save(file, overwrite=True)
sess.plot(block=False)
r = mne.io.read_raw(str(file))
r.plot(block=False)
raw.plot(block=False)
```

mscheltienne, 2021-08-24:
It looks like the problem is not in the saving part, but in the crop. I actually don't know why `sess.plot` shows the correct annotation.

```
raw.annotations.onset
Out[15]: array([ 8., 12., 18.])

sess.annotations.onset
Out[16]: array([ 8., 12., 18.])
```

`sess.annotations.onset` should be `[3, 7, 13]`. When calling `.crop()`, in `mne.io.base.py` line 1344 to 1347:

```
        if self.annotations.orig_time is None:
            self.annotations.onset -= tmin
        # now call setter to filter out annotations outside of interval
        self.set_annotations(self.annotations, False)
```

After the first line subtracting `tmin`, the onset is correct `array([ 3.,  7., 13.])`. After the setter, the onset is wrong at `array([ 8., 12., 18.])`. The setter adds the first time on line 693: `new_annotations.onset += self._first_time`.

agramfort, 2021-08-24:
I think this bug has been reported in the past and I suggested a temporary
fix
which is to set a meas_date so it's not None....

>

drammock, 2021-09-01:
> I don't get why `first_samp` might be different from 0 

This is a "feature" of Neuromag systems, where there is a distinction between "starting the acquisition system" and "starting the recording"

> and I don't get to what the annotations onsets are related to (i.e. what is the origin).

the origin is the beginning of the raw object instance, regardless of what it's `first_samp` value is.

mscheltienne, 2021-09-01:
> the origin is the beginning of the raw object instance, regardless of what its `first_samp `value is.

So would that line be the culprit? @693 in mne.io.Base.py

`new_annotations.onset += self._first_time`

mscheltienne, 2021-09-02:
@agramfort If I'm not opening a PR yet, it's because I know this is not the only culprit, and I can't really figure out what else is messing up the annotations. Let's take your example, with slightly renamed variables.

```
from datetime import datetime, timezone

import numpy as np
import mne

CH_NUM = 32
FS = 160

data = np.zeros((CH_NUM, FS * 20))
ch_names = [f'EEG{i+1}' for i in range(CH_NUM)]
ch_types = ['eeg'] * len(ch_names)
onset = [8, 12, 18]
duration = [2] * len(onset)
description = list(range(len(onset)))

info = mne.create_info(ch_names, ch_types=ch_types, sfreq=FS)
raw = mne.io.RawArray(data, info)
annotation = mne.Annotations(onset, duration, description)
raw = raw.set_annotations(annotation)

# raw.set_meas_date(datetime.now(tz=timezone.utc))

raw_copied = raw.copy()
raw_copied.crop(5, 18)
file = 'test_file_raw.fif'
raw_copied.save(file, overwrite=True)
raw_loaded = mne.io.read_raw(str(file))

raw.plot(block=False)
raw_copied.plot(block=False)
raw_loaded.plot(block=False)
```

If the measurement date is set, with the current code we get:

```
raw.annotations.onset
Out[2]: array([ 8., 12., 18.])

raw_copied.annotations.onset
Out[3]: array([ 8., 12., 18.])

raw_loaded.annotations.onset
Out[4]: array([ 8., 12., 18.])
```

@drammock Already, this does not make sense to me. If the annotations should be relative to the beginning of the raw object instance, then the timings for `raw_copied` and `raw_loaded` should be `-5`.

Without the measurement date, with the current code we get:

```
raw.annotations.onset
Out[6]: array([ 8., 12., 18.])

raw_copied.annotations.onset
Out[7]: array([ 8., 12., 18.])

raw_loaded.annotations.onset
Out[8]: array([13., 17.])
```

With a wrong plot for `raw_loaded`. That tells me that either `raw.save()` or `mne.io.read_raw` runs the annotations setter, adding a second time the variable `self._first_time` (5 seconds).

And finally, if we comment the line 693 in `mne.io.base.py` and thus don't add `self._first_time`:

```
raw.annotations.onset
Out[10]: array([ 8., 12., 18.])

raw_copied.annotations.onset
Out[11]: array([ 3.,  7., 13.])

raw_loaded.annotations.onset
Out[12]: array([ 3.,  7., 13.])
```

This is actually making sense to me. Both `raw_copied` and `raw_loaded` have the same onset, relative to the beginning of the instance. The durations are also similar and make sense:

```
raw.annotations.duration
Out[14]: array([2., 2., 2.])

raw_copied.annotations.duration
Out[15]: array([2.     , 2.     , 0.00625])

raw_loaded.annotations.duration
Out[16]: array([2.        , 2.        , 0.00625038])
```

However, this is the plot obtained:

[image]

I do believe this line 693 in `mne.io.base.py` is a mistake and should be removed, but something else is messing up the annotations.

mscheltienne, 2021-10-01:
A few words following discord discussion. The assumption is that:

- Annotation onset should be relative to the beginning of the data reflected by `first_samp`. -> Different from @drammock comment: `the origin is the beginning of the raw object instance, regardless of what its first_samp value is.`
- With this assumption, the problem would be limited to the I/O roundtrip.

I'll have a look when I have time.

PR #10040 (2021-12-13): Fix annotation cropping and I/O roundtrip
PR to fix #9527 

WIP:
- [x] Test to expose failure
- [x] Fix
closed 2021-12-13 by larsoner