mne-tools/mne-python #9536: spatial_colors as colorbar in image plots

transpersonify, 2021-07-07:
In evoked.plot() or evoked.plot_joint(), when spatial_colors=True, one can look at the channel locations. Is there an easy way to get colors of these channels and plot it as a colorbar alongside the Evoked image ? ( as plotted by evoked.plot_image() )

In any case, which colormap is used for spatial_colors? and is there an easy way to map it to the channel order?

mmagnuski, 2021-07-08:
@transpersonify The channel colors are mapped to channel position, not channel order. You can find the part of the code that does it here:
https://github.com/mne-tools/mne-python/blob/720a95c7599d933526b5db7fae823e4349176a01/mne/viz/evoked.py#L440

But you can just use the `_rgb` function that gets x, y, z channel positions, maps these values to 0 - 255 range and outputs as RGB matrix. So in short x position becomes red channel (so the more on the right, the more red), y -> G (more green towards front), z -> B (more blue towards top).

> Is there an easy way to get colors of these channels and plot it as a colorbar alongside the Evoked image ?

I'm not sure how the colorbar should look. Three separate colorbars for each (x, y, z) axis?

transpersonify, 2021-07-19:
@mmagnuski  Thank you for your answer! I did the dirty work and here is how the colorbar looks for the GSN-128 cap when I sort them by channel names:   https://pasteboard.co/KbRIK0H.png 
But You are right, maybe it is not the best representation, I will close the issue, thanks!

closed 2021-07-19 by transpersonify