mne-tools/mne-python #9546: viz.plot_alignment()'s "R" key binding doesn't fully reset the scene

hoechenberger, 2021-07-09:
Bug:
I was told that pressing the `R` key in the figure created by `plot_alignment()` is supposed to reset the camera / head to its initial position. However, it appears to only move the camera to a pre-defined distance from the center (which is not equal to the **initial** distance when the figure is first created), but doesn't reset any rotation, which is what I'm usually most concerned about.

Feature request:
Also I didn't know there were even keyboard shortcuts until @agramfort told me so. Would be nice to make this more discoverable – thinking of something like the toolbar we have for `Brain`, but simply adding some text, e.g., "Press R to reset the view" would already be very helpful.

cc @GuillaumeFavelier

GuillaumeFavelier, 2021-07-09:
AFAIK there is no such binding available in the standard 3d plots, at least not provided by MNE. So I think those shortcuts are probably offered by `pyvista` by default: 

https://docs.pyvista.org/plotting/plotting.html

I could disable the default ones to avoid the confusion and maybe use this issue as a feature request to provide a set of useful ones for us, what do you say?

hoechenberger, 2021-07-09:
Yes, I'd love that! Would it be difficult to add a tiny toolbar too?
But like I said, already having a simply "reset" key and an indicator on the screen letting you know **which** key it is would be super helpful!

GuillaumeFavelier, 2021-07-13:
> Yes, I'd love that! Would it be difficult to add a tiny toolbar too?

Yeah of course, with a button to reset the scene. But AFAIK, the concept of "scene" or "fig" is not clearly described in `mne-python` though.