mne-tools/mne-python #9548: ENH: [raw|epochs|evoked].get_data() specify tstart, tstop in seconds to get that segment

sappelhoff, 2021-07-09:
I often find myself wanting data from epochs in a certain time window, such as from -0.1 to 0.3 seconds. I don't want to look at `epochs.times` and compute the corresponding sample index every time, so I usually do something like `epochs.copy.crop(tmin, tmax).get_data()`. But copying the epochs is of course memory (and time) intensive.

I suggest we add two new parameters to the `get_data` method for raw, epochs, and evoked (evoked still needs the get_data method to be implemented, see #9244): `tstart` and `tstop` (or `tmin`, and `tmax`)

[The `get_data` method for raw](https://mne.tools/stable/generated/mne.io.Raw.html?highlight=get_data#mne.io.Raw.get_data) already has `start` and `stop` parameters, based on sample indices. The corresponding `tstart`/`tstop` params would operate the same way, but in seconds.

Adding something like `tstart` and `tstop` (in seconds) would also mean that we should add a `return_times=True|False` parameter to the [epochs.get_data](https://mne.tools/stable/generated/mne.Epochs.html?highlight=get_data#mne.Epochs.get_data) (and evoked.get_data, once we have it). That param is already there for raw.

Finally, maybe this would be a good opportunity to make the `get_data` interface equal between raw, epochs, and evoked.

epochs and evoked would then need to get the following that are already present in raw:

- units (#8473)
- start, stop (in samples)
- reject_by_annotation

What do you think?

larsoner, 2021-07-09:
Annotations are not preserved for epochs and evoked, so no reason to add them there.

I'd rather not add start/stop to epochs and evokeds. I kind of regret even having the sample-based one for raw -- I think it's mostly for backward compat, and it doesn't cost us much there so why not keep it. But I'd rather not add the sample-based stuff to more objects, I think it's not going to see much (if any) use. I'd rather showcase time-based limits in tutorials and examples, (but leave start/stop for raw for the reasons above).

Maybe we should use tmin/tmax rather than tstart/tstop? Not sure.

+1 for adding units to the epoch/evoked functions

drammock, 2021-07-09:
> I often find myself wanting data from epochs in a certain time window, such as from -0.1 to 0.3 seconds. I don't want to look at `epochs.times` and compute the corresponding sample index every time

Assuming you're aware of `epochs.time_as_index()`... is that the approach that you find too cumbersome?

> Maybe we should use tmin/tmax rather than tstart/tstop? Not sure.

+1 for `tmin` and `tmax`, and for addressing #8473 (units)

sappelhoff, 2021-07-09:
> Assuming you're aware of [epochs.time_as_index()](https://mne.tools/dev/generated/mne.Epochs.html#mne.Epochs.time_as_index)... is that the approach that you find too cumbersome?

thanks - I completely overlooked (forgot) about that method. So instead of `my_data = epochs.copy().crop(tmin, tmax).get_data()` I can do:

```python
idxs = epochs.time_as_index([tmin, tmax])
my_data = epochs.get_data()[..., idxs[0]:idxs[1]]
```

I actually think that's good enough and **would be happy to close this issue, unless** you (and others) think that it'd still be good to add `tmin` and `tmax` to `*.get_data()`.

units, and get_data for evoked are already tracked in their own issues, and the arguments not to add the other methods all make sense to me - thanks.

hoechenberger, 2021-07-10:
> **, unless** you (and others) think that it'd still be good to add `tmin` and `tmax` to `*.get_data()`.

I'd be in favor of having `tmin` and `tmax` params :)

rob-luke, 2021-07-10:
+1 for adding tmin and tmax. `epochs.get_data()[..., idxs[0]:idxs[1]]` is difficult to explain to a new user.

PR #9556 (2021-07-16): [MRG] ENH, FIX: Add tmin/tmax parameters to get_data methods, fix bug in find_bads_ecg
closes #9548 

~~**Work in progress**~~

To do

- [x] add to Raw
    - [x] add tests
- [x] add to Epochs
    - [x] pending #9553 to avoid conflicts
    - [x] add tests
- [x] add to Evoked
    - [x] pending #9555 because until then Evoked has no get_data method
    - [x] add tests
- [x] fix `find_bads_ecg`: https://github.com/mne-tools/mne-python/pull/9556#issuecomment-879106098
    - [x] add tests
- [x] add whatsnew (1) tmin/tmax, (2) fix of find_bads_ecg
closed 2021-07-16 by larsoner