mne-tools/mne-python #9551: Allow for rejection based on EMG PTP

hoechenberger, 2021-07-12:
Request from the forum:
https://mne.discourse.group/t/reject-epochs-based-on-peak-to-peak-signal-amplitude-ptp-of-emg

mscheltienne, 2021-07-15:
@hoechenberger I had a look to this as I thought this would be an easy first contribution to MNE. Isn't this already implemented?
I tested this loading a -raw file with 2 AUX channels, an EOG, and an ECG. I changed the type of the ECG to EMG to test.

```
import mne
import numpy as np

fname = r'path to -raw.fif'
raw = mne.io.read_raw(fname, preload=True)

mne.rename_channels(raw.info, {'AUX7': 'EOG', 'AUX8': 'EMG'}) 
raw.set_channel_types(mapping={'EMG':'emg', 'EOG':'eog'}) 

raw.set_eeg_reference(ref_channels='average', ch_type='eeg', projection=False)
raw.notch_filter(np.arange(50, 101, 50), picks=['eeg', 'emg', 'eog'], filter_length='auto', phase='zero')
mne.io.Raw.filter(raw, l_freq=1., h_freq=40., picks=['eeg'], method='iir',
                  iir_params=dict(order=4, ftype='butter', output='sos'))

events = mne.find_events(raw, stim_channel='TRIGGER')
reject = dict(eeg=80e-6,      # unit: V (EEG channels)
              emg=250e-6      # unit: V
              )

epochs = mne.Epochs(raw, events, event_id=4,
                    picks=['eeg', 'eog', 'emg'],
                    tmin=-0.2, tmax=0.8, reject=reject,
                    proj=True, baseline=(None, 0), preload=True)
```

And I do get a long verbose feedback in the console, with:

```
    Rejecting  epoch based on EMG : ['EMG']
    Rejecting  epoch based on EMG : ['EMG']
    Rejecting  epoch based on EEG : ['Fp2', 'F8', 'AF7', 'AF8']
    Rejecting  epoch based on EMG : ['EMG']
    Rejecting  epoch based on EMG : ['EMG']
    Rejecting  epoch based on EEG : ['Fp1', 'Fpz', 'Fp2', 'AF7', 'AF8']
    Rejecting  epoch based on EMG : ['EMG']
    Rejecting  epoch based on EMG : ['EMG']
```

`BaseEpochs._reject_setup` handles properly PTP rejection from any type of data as long as it is present in `self.info`, i.e. as long as it is present in the `picks` argument of `mne.Epochs`.

Did I oversee anything or is this actually already implemented?

agramfort, 2021-07-15:
it seems it's already working :)

>

GanshengT, 2021-07-17:
I have run a test on another dataset. Yes, Mathiue @mscheltienne you are right. PTP rejection works for  EMG data when creating epochs. In this case, I suggest updating the documentation page. In the following link https://mne.tools/stable/generated/mne.Epochs.html#, it is stated that valid keys are: 'grad', 'mag', 'eeg', 'eog', and 'ecg' for the **reject** parameter

hoechenberger, 2021-07-17:
Thanks for checking, @GanshengT. Would you like to do a PR? Such feedback and improvements from users are always highly welcome.

GanshengT, 2021-07-18:
The documentation is traced back to 
```
%(reject_epochs)s
```
in mne.Epochs.
Would you mind explaining what is the **%()s** structure and how is it related to the docstrings？

agramfort, 2021-07-18:
see here:

https://github.com/mne-tools/mne-python/blob/main/mne/utils/docs.py#L2301

to avoid duplicating all docstrings there is common dictionary and %()s is used to replace the key

mscheltienne, 2021-07-22:
One more idea. The current creation of epochs doesn't allow to pick channels X, Y from raw while rejecting based on channel Z. I think this is what created the confusion, the original post on the forum, and this issue. I guess the user wanted to create Epochs of e.g. EEG data while rejecting based on EMG. Currently, this would be a 2 step process since you would have to first create the epochs including the EMG, and then dropping the EMG channels.

Would it make sense to allow the user to reject based on the channels present in the provided raw instance instead of the channels present in the picks for the epochs?

agramfort, 2021-07-22:
I am not sure to follow. Standard practice for me in EEG or MEG studies is to have different channel types in one Epochs (eg MEG and EOG) and to reject full epochs based only on EOG.

drammock, 2021-07-22:
> I guess the user wanted to create Epochs of e.g. EEG data while rejecting based on EMG. Currently, this would be a 2 step process since you would have to first create the epochs including the EMG, and then drop the EMG channels.

I agree that it takes 2 steps, but that doesn't seem like too much to ask.  The second step is `epochs.pick('eeg')` to drop all other channel types, or `epochs.pick(['eeg', 'eog'])` to keep two types and drop all others... it's hard to imagine making it any easier.

> Would it make sense to allow the user to reject based on the channels present in the provided raw instance instead of the channels present in the picks for the epochs?

Imagine a case where the reject threshold for EEG is 40 μV, but the `picks` passed to the Epochs constructor only include EEG channels on the left side of the head.  Do we then reject an epoch if there are right-side channels that exceed 40 μV?  To me the answer is "obviously no".  The case of picking EEG and then using EMG for rejection is more debatable, but treating the different channel types differently becomes an implementation nightmare. For example, what if there are 2 EMG channels, and you want to use only one of them for rejection?  Right now the answer is simple: "only pick the one you want".  If the rejection code is allowed to consider channels that were not picked, then I'm not sure how you could handle that case.

mscheltienne, 2021-07-22:
@agramfort I was just trying to understand what caused the issue for the user. The practice I describe remains the same, but the final Epoch object created doesn't include the e.g. EOG channel. I was guessing that he tried to create epochs with a rejection on EMG while not keeping the EMG channel in the Epoch object, which failed because he did not have the EMG channel in the picks.

@drammock Your example is very good. Indeed, that would introduce even more problems and confusion for the users. The solution for a one-step process would be to allow in the rejection dictionary not only channel type to select all channels of a given type, but also channel names or even picks, but that could result in nasty rejection dictionary and code. The 2-step process looks good enough!

agramfort, 2021-07-23:
> @agramfort I was just trying to understand what caused the issue for the user. The practice I describe remains the same, but the final Epoch object created doesn't include the e.g. EOG channel. I [...]

ok makes sense. thx

PR #9589 (2021-07-24): change documentation about PTP rejection for diff ch types
#### Reference issue
Example: Fixes #9551.

#### What does this implement/fix?
Document specification about PTP rejection based on different types of channels
closed 2021-07-24 by hoechenberger