mne-tools/mne-python #9560: Make `mne.io.Raw` available as `mne.Raw`

cbrnr, 2021-07-12:
Two of the most important data classes, `mne.Epochs` and `mne.Evoked`, are available directly in `mne`. However, the class for continuous data is available as `mne.io.Raw` and not `mne.Raw` (because all `read_raw_*` readers also live in `mne.io`). For consistency, I propose to make `mne.io.Raw` available as `mne.Raw` (but keep `mne.io.Raw` for backward compatibility). WDYT?

larsoner, 2021-07-12:
-1 from me, I think if we think of `mne.io` as being for "raw IO" and all of our `mne.io.read_raw*` functions live there, the class should, too

agramfort, 2021-07-13:
-1 too

for the same reason. I also think our root namespace is already too big....

>

cbrnr, 2021-07-13:
+1 from me because IMO the three most important classes that represent data (`Raw`, `Epochs`, `Evoked`) should live together. @larsoner there is at least one `read_raw_*` function which produces an `Epochs` object (`read_raw_eeglab()`). @agramfort I'd rather move less important things into submodules and have these three classes in the root.

sappelhoff, 2021-07-13:
+1 because it seems natural to have the three main classes available from `mne`. @cbrnr's arguments make sense to me. But I won't personally spend time to get this done, so I guess I am a +0.5

drammock, 2021-07-13:
@cbrnr's argument makes sense to me too, but I agree with @agramfort that the root namespace is over-crowded.  I would not do this as a one-off change, but I would consider doing this as part of a larger namespace overhaul, done after careful consideration/planning, and with a longer-than-normal deprecation cycle.

larsoner, 2022-02-21:
Closing since I think we've converged on not changing this

closed 2022-02-21 by larsoner