mne-tools/mne-python #9563: DOC: Sphinx-gallery examples: replace bar of # characters with # %%

sappelhoff, 2021-07-13:
I suggest to replace the block of `#` characters with the more standard `# %%` in all sphinx-gallery examples.

I am talking about these blocks: https://github.com/mne-tools/mne-python/blob/308a0abf7c8b8b14c9afc6db2e3ed5b57787be31/tutorials/epochs/10_epochs_overview.py#L19-L23

The functionality for rendering the docs is equal, see: https://sphinx-gallery.github.io/stable/syntax.html#embed-rst-in-your-example-python-files

The advantage of using `# %%` is that many code editors recognize and parse `# %%` as "code blocks" so that you can execute an example similar to a notebook from your editor. This is good for playing around and debugging (using IPython). See the following quote from the docs (as linked above)

> The #%% and # %% syntax is consistent with the ‘code block’ (or ‘code cell’) separator syntax in Visual Studio Code Python extension, Visual Studio Python Tools, Jupytext, Pycharm Professional, Hydrogen plugin (for Atom) and Spyder. 

I switched to this style in many repos recently (e.g., mne-bids https://github.com/mne-tools/mne-bids/pull/830), and it's a good quality of live improvement for devs (to quickly try something) and users alike.

This could probably be done with a single search and replace operation. One thing that may be nice is to add a leading `# %%` **before** the imports, so that the imports become a code cell and can be executed. Currently they would not be parsed as a code cell, see the picture from above in a bigger view: https://github.com/mne-tools/mne-python/blob/308a0abf7c8b8b14c9afc6db2e3ed5b57787be31/tutorials/epochs/10_epochs_overview.py#L15-L23

--> code blocks would only start after the first block of `#` and that's undesirable.

What do you think?

drammock, 2021-07-13:
no strong objection.  Personally I find the full-79-char-width row to be useful as visual landmarks in the source, but I am convinced by the advantages you're pointing out. Would it work to do 
```
# %%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%
```
or 
```
# %% ##########################################################################
```
to get the best of both worlds?

hoechenberger, 2021-07-14:
+1000 for the suggestion.

Personally I don't like the visual appeal of:

> ```
> # %% ##########################################################################
> ```

Any editor / IDE recognizing the `# %%` as a cell marker will give a clear visual indication of where the cell starts and ends:

[image]

I wouldn't veto the 79-char proposal either; however if you asked my to choose, I'd pick `# %%`.

sappelhoff, 2021-07-14:
I also think `# %%` is enough, but I checked and it seems like the suggested `# %% ##########...` **would** work with sphinx-gallery (and it also works with code editors, I tested with VS Code):

```
    pattern = re.compile(
        r'(?P<header_line>^#{20,}.*|^# ?%%.*)\s(?P<text_content>(?:^#.*\s?)*)',
        flags=re.M)
```

Note the `^# ?%%.*`, source: https://github.com/sphinx-gallery/sphinx-gallery/blob/9d14861534f6bf0def4fd40f1148e12cf8f07a81/sphinx_gallery/py_source_parser.py#L182-L184

I think Dan can decide this one, as I would profit either way (`# %% #######...` vs bare `# %%`)

PR #9569 (2021-07-16): [MRG] DOC, Style: Use editor compatible code cells in gallery
closes #9563 

This is a global search and replace through the examples/ and tutorials/ directories for going from 79 `"#"` to `"# %% ####..."`.

I furthermore changed some licenses from "BSD style" to "BSD (3-clause)" (see 61d3f54f7b9c879fd506d660e80491c17d952e60) AND added licenses as follows to examples/tutorials that had no license:

```
# Authors: MNE-Developers
#
# License: BSD (3-clause)
```

(see a462d2e44d9839bdeaccc678823b0669bcff051f)

I then made use of the fact that each gallery file has a string `# License: BSD (3-clau [... cut]
closed 2021-07-16 by hoechenberger