mne-tools/mne-python #9566: Enable exporting EDF files

adam2392, 2021-07-13:
#### Describe the new feature or enhancement
There are gists available for using PyEDFlib to write EDF files, and pyedflib seems to be relatively well maintained and used.

This proposal is to enable exporting EDF files, with a warning of course that there maybe loss of data in the `info` object. This would also enable mne-bids to perform writing EDF file datasets, which is a RECOMMENDED file format for storage of EEG/iEEG data.

In addition, this would enable writing to an open-access file format that is heavily interfaced in EEG industry. For example, Persyst, an EEG-analytics company can only read in manufacturing formats (natus/NK/etc.) and EDF. It cannot read fiff, BrainVision, or other open source formats.

#### Describe your proposed implementation
This would extend https://github.com/mne-tools/mne-python/pull/9192 to add EDF file exporting. This would add an optional dependency with `pyedflib` (https://github.com/holgern/pyedflib).

#### Additional comments
Related issue in `mne-bids`: https://github.com/mne-tools/mne-bids/issues/840

larsoner, 2021-07-13:
Sounds reasonable to me

sappelhoff, 2021-07-14:
Just so that this doesn't get lost, @teuniz has chimed in in https://github.com/mne-tools/mne-bids/issues/840#issuecomment-879635128 and suggested https://gitlab.com/Teuniz/EDFlib-Python as an alternative.

Pros over `PyEDFlib`:
- developed/maintained by the same person who develops EDFlib (which pyEDFlib uses)

Cons: 
- no pypi package?
- documentation/tests not very mature (no info on supported py versions, no CI, no API docs)
- smaller user base (maybe hard to judge, I just see the GitLab stars)

Finally, I think both packages have an edge over https://github.com/bids-standard/pyedf which is a separate package independent of EDFlib, ... but it never took off.

cbrnr, 2021-07-14:
Thanks for the heads up @sappelhoff! IMO PyPI availability (pyEDFlib) trumps pure Python (EDFlib-Python) in this case, also because it is an optional dependency. Another issue might be the [custom license](https://gitlab.com/Teuniz/EDFlib-Python/-/blob/master/LICENSE), which is probably compatible with BSD, but I'm not sure. However, it shouldn't be too difficult to switch to a different EDF writer if the situation changes.

agramfort, 2021-07-15:
honestly it's really a bummer that there is 3 EDF libraries in Python. Can
we make these people
collaborate so there is one good, well maintained and well tested library
with a permissive BSD/MIT license?
I think it's necessary for the community to consolidate and the python
ecosystem has understood this for years
(contrarily to C++ or Javascripts world which are a lot more messy to
navigate).

my 2c

>

cbrnr, 2021-07-15:
If you count our own reader and [Biosig](https://pypi.org/project/Biosig/) there are even more. I completely agree that there should only be one well maintained EDF package in pure Python licensed under BSD/MIT.

I just noticed that [EDFlib-Python](https://www.teuniz.net/edflib_python/) is advertised as licensed under BSD-3 (in contrast to the `LICENSE` file in the repo). So maybe that's a good starting point? How would you proceed @agramfort?

Teuniz, 2021-07-15:
> If you count our own reader and [Biosig](https://pypi.org/project/Biosig/) there are even more. I completely agree that there should only be one well maintained EDF package in pure Python licensed [...]

Correct me if I'm wrong but the license file at https://gitlab.com/Teuniz/EDFlib-Python/-/blob/master/LICENSE has the same text as here: https://opensource.org/licenses/BSD-3-Clause

Teuniz, 2021-07-15:
> honestly it's really a bummer that there is 3 EDF libraries in Python. Can we make these people collaborate so there is one good, well maintained and well tested library with a permissive BSD/MIT license?

There can be many reasons why people sometimes don't want to collaborate.
One example: how strict should an EDF-reader be in (dis)allowing to read from a not 100% compliant EDF file?
EDFlib and EDFbrowser are very strict. Not everybody likes that.
For me, the longterm future of the EDF format is more important than trying to please angry or disappointed users who have a database full of non-compliant files.

cbrnr, 2021-07-15:
> Correct me if I'm wrong but the license file at https://gitlab.com/Teuniz/EDFlib-Python/-/blob/master/LICENSE has the same text as here: https://opensource.org/licenses/BSD-3-Clause

You're right, I didn't read the text but assumed because it is classified as "Other" that it is not any of the popular licenses (maybe it is possible to change this setting in GitLab somewhere?).

cbrnr, 2021-07-15:
> There can be many reasons why people sometimes don't want to collaborate.
One example: how strict should an EDF-reader be in (dis)allowing to read from a not 100% compliant EDF file?
EDFlib and EDFbrowser are very strict. Not everybody likes that.
For me, the longterm future of the EDF format is more important than trying to please angry or disappointed users who have a database full of non-compliant files.

Good example, we regularly deal with non-compliant files. I agree that it is important to enforce the file format, but often it's just tiny inconsistencies (such as different encodings) that are pretty easy to fix. This allows people to access their data, whereas the alternative is to have no data at all.

I think a good solution is to enforce strict mode by default, but allow some minor deviations optionally.

Regarding EDFlib-Python, would you be up for a PyPI release?

agramfort, 2021-07-15:
I would favor a strict support of the standard (whatever it means...)
otherwise you put the burden on yourself and not on the users who are often
very creative....

>

adam2392, 2021-07-15:
> Regarding EDFlib-Python, would you be up for a PyPI release?

+1 for using @Teuniz 's "strict" EDFlib-Python library. I'd rather have an EDF file that can be read by EDFBrowser, then just "EDF by name" and can only be read by MNE/EEGlab functions.

cbrnr, 2021-07-15:
An EDF writer should definitely be super strict. My comments on relaxing some restrictions a little apply only to a reader.

Teuniz, 2021-07-16:
> An EDF writer should definitely be super strict. My comments on relaxing some restrictions a little apply only to a reader.

That's not going to happen. EDFlib is finished and offered "as is". Consider it an LTS release.  
Any bugs wiil be fixed ofcourse. New function calls will be taken into consideration if it makes sense.  
API breakage is not going to happen though.  
Ofcourse, there's always the option to clone & fork and turn it into a "community project" but my support will be limited  
to the vanilla EDFlib.

cbrnr, 2021-07-16:
I think we were mostly talking about the writer component, because we already have our own reader which is not that strict.

Teuniz, 2021-07-16:
Understood and no problem. The Python version of EDFlib is split into a reader and a writer part  
(they are different files) and you can simply grab the writer only and don't bother with the reader if you wish.

adam2392, 2021-07-16:
Awesome!

What would be the next best course of action here if EDFlib isn't pypi released?

drammock, 2021-07-16:
@Teuniz would you be open to doing a PyPi release of python EDFlib?  One of us could help out with the changes needed for packaging, if that makes a difference.  If not, I think our only option for using EDFlib is incoporating the writer into our `externals/` (which we tend to be hesitant to add things to)

Teuniz, 2021-07-16:
> @Teuniz would you be open to doing a PyPi release of python EDFlib? One of us could help out with the changes needed for packaging, if that makes a difference. If not, I think our only option for [...]
I'll look into it, I have never done it before but I have a colleague with lots of Python experience.
It's just that we are entering the holiday period here, so it will take some time.

adam2392, 2021-08-04:
I took a stab at sketching out the export functionality. The barebones works (with some details not matching up).

Just a friendly ping to see if there was any update on getting a PyPi release? I can help with it if needed?

Teuniz, 2021-08-05:
> Just a friendly ping to see if there was any update on getting a PyPi release?

It's done, you can find it here: https://pypi.org/project/EDFlib-Python/

PR #9643 (2021-08-24): [MRG] Enable export of EDF files
#### Reference issue
Closes: #9566 

#### What does this implement/fix?
Enables exporting of EDF+ ~~and BDF files,~~ which support annotations.

Note that the montage (i.e. info['dig']) is technically lost upon export.

#### Additional information
If you look at the Jupyter notebook, then I'm not getting lossless conversion. I think I am perhaps setting the physical max/min and digital max/min incorrectly.
closed 2021-08-24 by cbrnr