mne-tools/mne-python #9576: find_events is empty for EDF file and annotation channel can't be loaded

mkoohim, 2021-07-16:
I plan to use depression EEG database from the following resource:
https://figshare.com/articles/dataset/EEG_Data_New/4244171

It is in EDF format and the last channel is annotation with "EDF Annotations" label. 

I tried to one of the file with the annotation, but unfortunately I can't load the annotation:

`raw = mne.io.read_raw_edf('H S30 EC.edf', preload=True)`
`print(raw.info["ch_names"])`

The output chnnels are:

`['EEG Fp1-LE', 'EEG F3-LE', 'EEG C3-LE', 'EEG P3-LE', 'EEG O1-LE', 'EEG F7-LE', 'EEG T3-LE', 'EEG T5-LE', 'EEG Fz-LE', 'EEG Fp2-LE', 'EEG F4-LE', 'EEG C4-LE', 'EEG P4-LE', 'EEG O2-LE', 'EEG F8-LE', 'EEG T4-LE', 'EEG T6-LE', 'EEG Cz-LE', 'EEG Pz-LE', 'EEG A2-A1']`

which is not included annotation channel. I also tried:

`raw = mne.io.read_raw_edf('H S30 EC.edf', stim_channel=-1, preload=True)`

`raw = mne.io.read_raw_edf('H S30 EC.edf', stim_channel='auto', preload=True)`

And the results was same. Actually my final goal is extracting events using `mne.find_events` but first I need to load annotation.

Any solution to extract events from these EDF files? Thanks.

cbrnr, 2021-07-16:
I checked this file with [EDFbrowser](https://www.teuniz.net/edfbrowser/) and [SigViewer](https://github.com/cbrnr/sigviewer), and it really seems like the annotation channel (although present) does not contain any annotations.

mkoohim, 2021-07-16:
Ok. Thanks. So the problem is rooted into the EDF files of this database.

closed 2021-07-16 by drammock