mne-tools/mne-python #9577: [BUG] `mne.label.find_pos_in_annot` doesn't check `coord_frame`

alexrockhill, 2021-07-16:
It doesn't seem like a good idea to allow any position to be passed to `find_pos_in_annot` without checking that it is in the `mri` coordinate frame. If someone has the wrong coordinate frame, the error will be silent and they will get the wrong ROI which to me seems like a pretty good risk for incorrect results that could be avoided by enforcing the coordinate frame or maybe rather making this a class function of `Dipole`.

larsoner, 2021-07-16:
I don't think `Dipole` would be the right subclass, in general the point you're talking about could be anything (iEEG sensor, part of a mesh, an elementary current dipole, etc.). I think we should do two things:

1. Document clearly that if an `ndarray` or list-like is passed, these must be in the MRI coordinate frame
2. Add support for other objects as we need them, such as DigPoint instances (?), where we can check the coordinate frame

FWIW `Dipole` always lives in the *head* coordinate frame, so if we wanted to add support for that in (2) then we'd have to have a `trans` argument in that function, if we don't already.

But really I think we should do the doc update (1) first and then let actual use cases guide (2)

alexrockhill, 2021-07-16:
Maybe it should be in a mixin for things with positions?

larsoner, 2021-07-16:
What objects + use cases do you have in mind?

alexrockhill, 2021-07-16:
I think just `DigMontage` and `Dipole` unless you know more that have positions... now that I say that it might be better just to make an ``inst`` that takes either

larsoner, 2021-07-16:
`Dipole` is always in head coords and I don't think it's a common use case anyway. Let's just start from `montage` (and not worry about Dipole) since it's the use case we actually want to support for iEEG?

closed 2021-07-19 by larsoner