mne-tools/mne-python #9581: Annotations.delete requires Numpy 1.19 or above

rob-luke, 2021-07-17:
#### Describe the bug
This block of code in the tutorials produces the wrong result when used with Numpy 1.18

https://github.com/mne-tools/mne-python/blob/4875e04cdebba132fc637ca65b3cb50584aee8b7/tutorials/preprocessing/70_fnirs_processing.py#L41-L46

#### Steps to reproduce
Run the https://github.com/mne-tools/mne-python/blob/main/tutorials/preprocessing/70_fnirs_processing.py tutorial with Numpy 1.18

#### Expected results
What is displayed here https://mne.tools/dev/auto_tutorials/preprocessing/70_fnirs_processing.html#providing-more-meaningful-annotation-information

#### Actual results
Instead of removing the two annotations with trigger code 15.0, one 15.0 is remove and one other condition is removed.

#### Additional information
I think this is due to a change in how np.delete works that occurred in 1.19. See https://numpy.org/doc/stable/release/1.19.0-notes.html#expired-deprecations

#### Question

What should I do?

1. Do we place a minimum version requirement on Numpy? Currently its v1.15 https://github.com/mne-tools/mne-python/blob/4875e04cdebba132fc637ca65b3cb50584aee8b7/requirements.txt#L2
2. Change the internal function of annotations.remove?
3. Other option?

larsoner, 2021-07-19:
> Do we place a minimum version requirement on Numpy? Currently its v1.15

At some point we settled on supporting the last two years of our dependencies in each release, and we [bump the requirements only at release time (see step 18)](https://github.com/mne-tools/mne-python/wiki/How-to-make-a-release)

> Change the internal function of annotations.remove?

Usually what we try to do is copy-paste a working version (can be from NumPy `main` if you want) in this case of `np.delete` into `mne/fixes.py` (with a comment about what minimum version of NumPy allows us to stop using it) and use that in `annotations.delete`. In the unfortunate case that `np.delete` is in C, then we'll have to write our own little wrapper and hope it's functionally equivalent, but hopefully that won't be the case!

rob-luke, 2021-07-21:
Thanks for the explanation. Ill find time for this

drammock, 2021-10-12:
This is actually not a bug in MNE-Python; docstring of annotations.delete says it takes `int` or array-like of `int`, so passing it an array of booleans (as is done in the tutorial) is the problem.  I'll open a PR to fix.