mne-tools/mne-python #9584: [BUG] `mne.viz.Brain` raises cryptic error when plotting 'seghead', works for `mne.viz.plot_alignment`

alexrockhill, 2021-07-18:
Here's some minimally reproducible code:

```
import os.path as op
from mne.datasets import sample
import mne

data_path = sample.data_path()
sample_dir = op.join(data_path, 'MEG', 'sample')
subjects_dir = op.join(data_path, 'subjects')
evoked = mne.read_evokeds(op.join(sample_dir, 'sample_audvis-ave.fif'))[0]
trans = mne.read_trans(op.join(sample_dir, 'sample_audvis_raw-trans.fif'))
mne.viz.plot_alignment(evoked.info, trans, subject='sample', subjects_dir=subjects_dir, surfaces=['seghead'])  # works
mne.viz.Brain('sample', 'lh', 'seghead', subjects_dir=subjects_dir)  # raises error
```

Stack trace
```
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/Users/alexrockhill/projects/mne-python/mne/viz/_brain/_brain.py", line 523, in __init__
    mesh.add_overlay(
  File "/Users/alexrockhill/projects/mne-python/mne/viz/_brain/_brain.py", line 178, in add_overlay
    self._apply()
  File "/Users/alexrockhill/projects/mne-python/mne/viz/_brain/_brain.py", line 194, in _apply
    self._renderer._set_mesh_scalars(
  File "/Users/alexrockhill/projects/mne-python/mne/viz/backends/_pyvista.py", line 699, in _set_mesh_scalars
    mesh.point_arrays[name] = scalars
  File "/Users/alexrockhill/software/anaconda3/envs/mne/lib/python3.9/site-packages/pyvista/core/datasetattributes.py", line 57, in __setitem__
    self.append(narray=value, name=key)
  File "/Users/alexrockhill/software/anaconda3/envs/mne/lib/python3.9/site-packages/pyvista/core/datasetattributes.py", line 211, in append
    raise ValueError(f'narray length of ({narray.shape[0]}) != required length ({array_len})')
ValueError: narray length of (155407) != required length (267122)
```

larsoner, 2021-07-18:
seghead is not a valid brain surface, so I doubt baby folks will hit this. We should maybe just add a "add_head" function that can use a surface name or BEM

alexrockhill, 2021-07-18:
Ok so I guess it should raise a more informative error in this case? `surf/lh.seghead` is the same kind of surface as all the rest of the surfaces AFAIK so it might make sense to have this be plottable by the same API alternately

larsoner, 2021-07-18:
> surf/lh.seghead is the same kind of surface as all the rest of the surfaces AFAIK so it might make sense to have this be plottable by the same API alternately

I think it's not the same actually. The file format is the same but it's not a left hemisphere surface (and does not share an identical triangulation, which all the other sphere/pial/white etc. surfaces do) so we should not treat it the same

alexrockhill, 2021-07-18:
> I think it's not the same actually. The file format is the same but it's not a left hemisphere surface (and does not share an identical triangulation, which all the other sphere/pial/white etc. surfaces do) so we should not treat it the same

Ah okay good to know! I can write a quick PR to raise an error. Are there any other surfaces that are formatted as lh.? that also don't work?

agramfort, 2021-07-19:
+1 for a new head parameter. Maybe just head=True | False | str ?  do we need the str option?

larsoner, 2021-07-19:
I would really rather not add a new parameter to the Brain constructor, the signature is already too long. Plus there are options one might want to specify (which surface/density/filename, color, opacity) so a dedicated add_head method seems better to me 

> Are there any other surfaces that are formatted as lh.? that also don't work

I would raise an error for surface=seghead, I'm not sure if there are other surfaces that are incompatible

alexrockhill, 2021-07-19:
> a dedicated add_head method seems better to me

I really like the `add_XXX` methods, they seem easy to use and understand and easy to maintain as they are a lot simpler. I think eventually they might all need a `remove_XXX` that will complicate things a bit but that is a +1 from me for style.

larsoner, 2021-07-19:
> I think eventually they might all need a remove_XXX that will complicate things a bit but that is a +1 from me for style.

Can you open an issue for this and ping @GuillaumeFavelier to consider after the `coreg` class refactor is done at some point? One way to make this simple is to have each method return a list of VTK actors, and the `Brain` instance can keep track internally of a list of actors it has added. Then we can just have a `brain.remove(actors)` method that `.pop`s each actor from the internal list and removes it from each plotter.

PR #9618 (2021-07-29): [ENH, MRG] Add `mne.viz.Brain.add_head` method to display head
Fixes #9584.
closed 2021-07-29 by larsoner