mne-tools/mne-python #9606: Better error message for 3D plots when pyvistaqt is missing

cbrnr, 2021-07-23:
When creating a 3D plot with e.g. `mne.viz.plot_alignment` (not sure if there are any additional similar functions), I get an error if `pyvistaqt` is not installed:

```
RuntimeError: Could not load any valid 3D backend: pyvista, mayavi, notebook
```

Note that I actually have `pyvista` installed, but it seems like `pyvistaqt` is *also* required. A simple solution would be to update the error message like this (since `pyvistaqt` depends on `pyvista`):

```
RuntimeError: Could not load any valid 3D backend: pyvistaqt, mayavi, notebook
```

hoechenberger, 2021-07-23:
cc @GuillaumeFavelier

GuillaumeFavelier, 2021-07-23:
Hm... I agree it's better.

GuillaumeFavelier, 2021-07-23:
I'll open a PR to rename the backend

closed 2021-07-26 by hoechenberger