mne-tools/mne-python #9609: Axial to planar gradiometer transformation

PiusKern, 2021-07-23:
For axial gradiometer data, e.g. from CTF 275 systems, a transformation to virtual planar gradiometers would be helpful. This is a standard procedure in many data analysis pipelines for axial systems. It would facilitate plotting and subsequent interpretation.  

The issue has been briefly discussed during the office hour on 2021-07-23.

@britta-wstnr

britta-wstnr, 2021-07-23:
Thanks @PiusKern. 
I agree, some methods for this would be useful. During the office hours, we also briefly touched upon how to solve this computationally. As discussed, I will check in detail what FieldTrip does and report back, then we can discuss how we want to proceed.

britta-wstnr, 2021-07-28:
FieldTrip supports different methods on how to compute planar from axial grads: `sourceproject` using a minimum current estimate, `fitplane` using the neighborhood structure, and `sincos` using the neighborhood structure as well, but more flexibly than the `fitplane` option. `sincos` is default here.

For the combination of the vertical and horizontal parts, using an SVD is one option,  `sum` (`sqrt(horizontal ** 2 + vertical ** 2)`) and `complex` are others.

larsoner, 2021-07-28:
I would start with just the `sourceproject` method since we have it implemented already and I think @agramfort suggested it should be the best in principle. We can assume YAGNI on the others until we see otherwise (especially if the values aren't that different, or they do differ but the sourceproject is better for example in round-trip test to a different system/sensor type and back).

britta-wstnr, 2021-07-28:
FWIW, there is some rather old opinion by Robert, referring to simulations, saying the `sincos` method outperforms the others. See here: https://mailman.science.ru.nl/pipermail/fieldtrip/2005-August/026154.html

But I agree, if it's there, let's start with it and then we can always add this (and compare again, if we are feeling adventurous :wink: )

britta-wstnr, 2021-08-18:
@larsoner What would be a good API for this / where should this live?

larsoner, 2021-08-18:
We have `inst.interpolate_bads` as a mixin for raw/epochs/evoked, I think it should be part of that same mixin as a new method:
```
def interpolate_to(self, sensors=None, dev_head_t=None):
```
where `sensors` can be (to start) only `None`, `'ctf planar grad'`, `'ctf'`, and `'neuromag'`? If `sensors is None` then it's useful just for transforming to a new `dev_head_t`.

I suggest just these three string options to start because in this issue `'planar grad'` is desired, and I could see people wanting to know what their data would look like if they had been acquired on these other systems. Having `ctf` and `neuromag` is also nice because we can round-trip from actual Neuromag data to CTF and back to Neuromag and check the agreement, same with CTF to Neuromag to CTF.

Under the hood this should use the same underlying machinery as `interpolate_bads` -- it's actually quite simple hopefully as we "just" need to set `info_to` properly by replacing some `info_from['chs']` with the known canonical ones for the given system.

I like `sensors` as the name because in theory we could eventually support remapping other channel types (e.g., EEG) someday. But for now the function should operate only on MEG data.

britta-wstnr, 2021-08-20:
Thanks for the detailed overview @larsoner ! I hope to give this a shot soon, maybe together with @PiusKern ?

larsoner, 2022-04-20:
I forgot we had come up with an API for this! Just one more hint to add to [what I wrote above](https://github.com/mne-tools/mne-python/issues/9609#issuecomment-901203391), I think internally you should "just" need an appropriate call to `_map_meg_or_eeg_channels` to do the heavy lifting here, as it's what's used by `interpolate_bads`.

britta-wstnr, 2022-04-20:
nice, thanks @larsoner !

finch-f, 2024-03-05:
Hi, I am wondering if there has been a function to transform axial gradiometers to virtual planar gradiometers? I have tried the as_type() function. However, it fails on my CTF275 data.

larsoner, 2024-03-05:
No it has not been implemented yet, still needs a volunteer to give it a shot

contsili, 2024-10-21:
Hi all, after a discussion with @britta-wstnr I would like to take over this issue. 

> We have `inst.interpolate_bads` as a mixin for raw/epochs/evoked, I think it should be part of that same mixin as a new method:
> 
> ```
> def interpolate_to(self, sensors=None, dev_head_t=None):
> ```
@larsoner do you still think this is still a good API?

Additionally, I see that you decided on `sourceproject` method. Shall we still proceed with this method?

contsili, 2024-12-17:
> I would start with just the `sourceproject` method since we have it implemented already and I think @agramfort suggested it should be the best in principle. We can assume YAGNI on the others until [...]

@larsoner, could you point me to where the `sourceproject `method is implemented in the mne repo?

britta-wstnr, 2025-02-05:
Hi @contsili! 👋  Are you still interested in tackling this? 🙂

contsili, 2025-02-05:
Hi @britta-wstnr.Yes I am still interested. I started working and I have less time but I will do it!

I think a good plan is to wait for this PR #13040 to be merged and then I can extend the functionality of `interpolate_to`

britta-wstnr, 2025-02-05:
Hi @contsili! Great to hear 🙌  
Don't hesitate to open a PR with snippets already, it's better to have early discussions on implementations (less work for you and the code reviewers! 🙂 ). You can open a PR in DRAFT mode to signal that this is not a fully-working implementation yet. Looking forward to it!