mne-tools/mne-python #9616: Couple issues with interactive plots when integrated into PyQt5

LMBooth, 2021-07-27:
#### Describe the bug
This issue is related to issue #8142 - I'm currently implementing MNE 0.24dev0 (having the same issues with 0.23 too) into a pyqt5 application and am trying to mark channels as bad from the plot, however, the bads field in raw.info is not updating (i suspect it may be related to me having show set to False in order to put the figure in a canvas). The tutorial for [handling bad channels](https://mne.tools/stable/auto_tutorials/preprocessing/15_handling_bad_channels.html) states block should = True for the bads field to update automatically when the plot is interacted with. Please find a minimum working example below:

#### Steps to reproduce
```python
from PyQt5 import QtWidgets, QtGui
import matplotlib
from matplotlib.backends.backend_qt5agg import FigureCanvasQTAgg as FigureCanvas
from matplotlib.backends.qt_compat import QtCore
import sys
from mne.datasets import sample
import mne
import os 
class My_Main_window(QtWidgets.QMainWindow):
    def __init__(self, parent=None):
        super(My_Main_window, self).__init__(parent)
        
        sample_data_folder = sample.data_path()
        sample_data_raw_file = os.path.join(sample_data_folder, 'MEG', 'sample',
                                    'sample_audvis_filt-0-40_raw.fif')
        self.raw = mne.io.read_raw_fif(sample_data_raw_file)
        self.fig = self.raw.plot(show=False, block=True)
        
        self.canvas = FigureCanvas(self.fig)
        self.canvas.draw()
        layout = QtWidgets.QVBoxLayout()
        layout.addWidget(self.canvas)
        self.button = QtWidgets.QPushButton('Print Bad', self)
        self.button.clicked.connect(self.BadButton)
        layout.addWidget(self.button)
        
        self.stack = QtWidgets.QTabWidget()
        self.stack.setLayout(layout)
        self.setCentralWidget(self.stack)

        # Give the keyboard focus to the figure instead of the manager:
        # StrongFocus accepts both tab and click to focus and will enable the
        # canvas to process event without clicking.
        # https://doc.qt.io/qt-5/qt.html#FocusPolicy-enum
        self.canvas.setFocusPolicy(QtCore.Qt.StrongFocus | QtCore.Qt.WheelFocus)
        self.canvas.setFocus()

    def BadButton(self):
        print(self.raw.info['bads'])

if __name__ == '__main__':
    app = QtWidgets.QApplication(sys.argv)
    app.setApplicationName("Plot")
    main_win = My_Main_window()
    main_win.show()
    app.exec()
```

#### Expected results
After pressing the "Print Bad" button i expect any selected channels from the plot to print to the terminal.

#### Actual results
 Only the preset bad channels (['MEG 2443', 'EEG 053']) are printed from the sample_audvis_filt-0-40_raw.fif sample dataset.

#### Additional problem which may be related
Another issue i'm having relates to the psd topomaps when integrated in to pyqt, namely all the previously hidden plot figures appear when selecting a frequency band to plot. 

To replicate this you can change the line `self.fig = self.raw.plot(show=False, block=True)` to `self.fig = self.raw.plot_psd(show=False)` then click and drag over a frequency band on the psd plot.

Any comments or proposed workarounds would be greatly appreciated.

Kind regards,

Liam

LMBooth, 2021-09-02:
#9687 by @marsipu could provide a fix for the issue, so i'll hold off working on a fix for now.

cbrnr, 2021-09-15:
I'm not sure if the documentation is correct, but we've discussed this issue before (can't find it right now). I think we're currently updating the `bads` list only after closing the figure, but we should be able to do that on the fly. @drammock can you remember what we discussed?

LMBooth, 2021-09-15:
>  I think we're currently updating the `bads` list only after closing the figure

Thanks @cbrnr for pointing that out, it makes sense as to why it's not updating. 

Something else I noticed whilst embedding the plotter in to PyQt5 - opening popup figures from a displayed figure (topomaps from `plot_psd`) seems to open all previously blocked figures (`block=True`) too, is this likely related or a separate issue?

drammock, 2021-09-15:
> @drammock can you remember what we discussed?

I think it was agreed that updating bads in real time was a good change to make. I'll see if I can find time this week to do that.