mne-tools/mne-python #9627: [BUG] `mne.viz.plot_alignment` silently assumes unknown coordinate frame is in head

alexrockhill, 2021-07-30:
@larsoner suggested warning that the coordinate frame is unknown might be the best option but as in this example, an unknown coordinate frame is silently accepted as being in "head" coordinates:

```
import os.path as op
import numpy as np
import mne
data = np.random.normal(size=(16, 100))
ch_names = ['S1_D1 hbo', 'S1_D1 hbr', 'S2_D1 hbo', 'S2_D1 hbr',
            'S3_D1 hbo', 'S3_D1 hbr', 'S4_D1 hbo', 'S4_D1 hbr',
            'S5_D2 hbo', 'S5_D2 hbr', 'S6_D2 hbo', 'S6_D2 hbr',
            'S7_D2 hbo', 'S7_D2 hbr', 'S8_D2 hbo', 'S8_D2 hbr']
ch_types = ['hbo', 'hbr', 'hbo', 'hbr',
            'hbo', 'hbr', 'hbo', 'hbr',
            'hbo', 'hbr', 'hbo', 'hbr',
            'hbo', 'hbr', 'hbo', 'hbr']
sfreq = 10.  # in Hz
info = mne.create_info(ch_names=ch_names, ch_types=ch_types, sfreq=sfreq)
raw = mne.io.RawArray(data, info, verbose=True)

subjects_dir = mne.datasets.sample.data_path() + '/subjects'
mne.datasets.fetch_fsaverage(subjects_dir=subjects_dir)

montage = mne.channels.make_standard_montage('artinis-octamon')
raw.set_montage(montage)
trans = mne.channels.compute_native_head_t(montage)

fig = mne.viz.create_3d_figure(size=(800, 600), bgcolor='white')
fig = mne.viz.plot_alignment(
    raw.info, trans=trans, subject='fsaverage', subjects_dir=subjects_dir,
    surfaces=['brain', 'head'], coord_frame='mri', dig=True, show_axes=True,
    fnirs=['channels', 'pairs', 'sources', 'detectors'], fig=fig)
print({ch: raw.info['chs'][i]['coord_frame'] for i, ch in enumerate(raw.ch_names)})
```

Adapted from https://mne.tools/stable/auto_tutorials/io/30_reading_fnirs_data.html

alexrockhill, 2021-07-30:
Huh, `artinis-octamon` has fiducial points so there might also be an issue with the coordinate frame not being set to "head" properly...

rob-luke, 2021-07-30:
That example starts with "This method is not supported and users are discoraged to use it."

There was a lengthy discussion about this when the example was added (git blame if you're interested).

Are the _official_ fNIRS readers causing the same issue? If so I can modify them to specify the head frame.

alexrockhill, 2021-07-30:
I suspect this is a widespread, non-fNIRS specific issue as I ran into it for the stereoEEG tutorial as well

alexrockhill, 2021-07-30:
Ok, actually this looks like it is a fNIRS-specific issue. The following is correctly set to the "head" coordinate frame:

```
import numpy as np
from matplotlib import pyplot as plt

import mne

print(__doc__)

# %%
# Create fake data
# ----------------
#
# First we will create a simple evoked object with a single timepoint using
# biosemi 10-20 channel layout.

biosemi_montage = mne.channels.make_standard_montage('biosemi64')
n_channels = len(biosemi_montage.ch_names)
fake_info = mne.create_info(ch_names=biosemi_montage.ch_names, sfreq=250.,
                            ch_types='eeg')

rng = np.random.RandomState(0)
data = rng.normal(size=(n_channels, 1)) * 1e-6
fake_evoked = mne.EvokedArray(data, fake_info)
fake_evoked.set_montage(biosemi_montage)
print(fake_evoked.info['chs'][0])
```

Looks like a really simple fix though. Will push a PR asap.

alexrockhill, 2021-08-02:
I don't think this should be closed, this still happens and likely will until #9585 refactors `plot_alignment` and is merged.

PR #9630 (2021-07-31): [BUG, MRG] fix coordinate frame in fNIRS bug
Fixes #9627.
closed 2021-08-19 by alexrockhill