mne-tools/mne-python #9635: [BUG] Coordinate Frame not Saved

alexrockhill, 2021-08-02:
I'm not sure if I'm missing something obvious but it looks like the coordinate frame doesn't survive the round trip to and from disc. I'm not sure why that is.

```
import numpy as np
import mne

raw = mne.io.RawArray(
    np.random.random((10, 1000)),
    mne.create_info([f'ch{i}' for i in range(1, 11)],  1000, ['seeg'] * 10))
montage = mne.channels.make_dig_montage(
    {ch: np.random.random((3,)) for ch in raw.ch_names}, lpa=[1, 0, 0], nasion=[0, 0.1, 0], rpa=[-1, 0, 0],
    coord_frame='mri')
raw.set_montage(montage)
raw.save('tmp-raw.fif')
print(raw.info['chs'][0])  # coord_frame == 'head'
raw2 = mne.io.read_raw('tmp-raw.fif')
print(raw2.info['chs'][0])  # coord_frame == 'unknown'
```

alexrockhill, 2021-08-03:
To update, it looks like this was on purpose but I think it would be better to assume that `seeg`, `ecog` or `dbs` channel positions are in head coordinate frames because if they are set using `inst.set_montage(montage)` they will be converted. To my knowledge, there are no other public APIs to set the channel positions so it seems like they should be assumed to be in head if read from a disk.

alexrockhill, 2021-08-19:
@larsoner, this is in https://github.com/mne-tools/mne-python/pull/9585/, just wanted to make sure you were aware and thought this was reasonable

larsoner, 2021-08-19:
Yeah I think it's just an oversight that we never added these channel types to the inferred-coord-frame list

alexrockhill, 2021-08-19:
Great, just making sure it wasn't for a reason I wasn't thinking of

closed 2021-08-19 by alexrockhill