mne-tools/mne-python #9639: BEM changes depending on calling path

jstout211, 2021-08-03:
#### Describe the bug
The BEM model is different depending on where the the code is called from.  From what I can tell it appears to be missing the CRAS offset.  This is  in the viz.plot_aligment, but I have also seen it in the source localization outputs.

#### Steps to reproduce
``` 
import mne
import os, os.path as op

subjects_dir=op.join(os.environ['HOME'], 
                 'mne_data/MNE-somato-data/derivatives/freesurfer/subjects')

#In mne_data directory
os.chdir(subjects_dir)
mne.viz.plot_bem('01', subjects_dir=subjects_dir)

#In home directory
os.chdir(os.environ['HOME'])
mne.viz.plot_bem('01', subjects_dir=subjects_dir)
```

#### Expected results
[image]

#### Actual results
[image]

#### Additional information
Platform:      Linux-5.4.0-80-generic-x86_64-with-glibc2.10
Python:        3.8.8 (default, Feb 24 2021, 21:46:12)  [GCC 7.3.0]
Executable:    /home/jstout/miniconda3/envs/enigma/bin/python
CPU:           x86_64: 20 cores
Memory:        125.5 GB

mne:           0.23.0
numpy:         1.20.1 {blas=openblas, lapack=openblas}
scipy:         1.6.1
matplotlib:    3.3.4 {backend=module://ipykernel.pylab.backend_inline}

sklearn:       0.24.1
numba:         0.53.0
nibabel:       3.2.1
nilearn:       0.7.1
dipy:          1.4.0
cupy:          Not found
pandas:        1.2.3
mayavi:        Not found
pyvista:       0.29.0 {pyvistaqt=0.3.0, OpenGL 4.5.0 NVIDIA 460.91.03 via Quadro RTX 8000/PCIe/SSE2}
vtk:           9.0.1
PyQt5:         5.15.4

jstout211, 2021-08-04:
I am able to replicate this consistently on two machines (redhat and ubuntu), using different newly installed conda environments with mne 0.21-0.23 (0.20 seemed to work).  At the same time, I have had others in our group unable to replicate this issue on their machine.  I dont' know if this is a true issue or a configuration on my end.  I will keep looking into this.

jstout211, 2021-08-04:
I think that i have figured out the issue, but I am not sure what the preferred solution would be.  Basically during the plot_bem (and setup_volume_source_space), the command _check_mri will create a full path for the mri (from the subject + subj_dir ...) unless it finds the filename within the current folder.  The default mri name is T1.mgz for the function call.  I had a T1.mgz file in my home directory, so it was preferrentially using this file rather than the one in subjects_dir/subject/mri folder.  

Moving a T1.mgz file to/from the path of the function call has been able to create/fix the problem on different machines (linux + mac). 

def _check_mri(mri, subject, subjects_dir):
    _validate_type(mri, 'path-like', 'mri')
    if not op.isfile(mri):                    <<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<
        if subject is None:
            raise FileNotFoundError(
                'MRI file %r not found and no subject provided' % (mri,))
        subjects_dir = get_subjects_dir(subjects_dir, raise_error=True)
        mri = op.join(subjects_dir, subject, 'mri', mri)
        if not op.isfile(mri):
            raise FileNotFoundError('MRI file %r not found' % (mri,))
    return mri

agramfort, 2021-08-05:
Can you open a PR?

jstout211, 2021-08-05:
Yeah I will submit a PR by the end of the day

PR #9650 (2021-08-06): FIX: Remove ambiguity in local filename check
#### Reference issue
Fixes #9639 

#### What does this implement/fix?
Prior to this fix, the _check_mri would proceed if it found the default mri (T1.mgz).  If the default filename happened to be in the calling directory, it would be returned from the function rather than performing the expected search of the freesurfer folder.

The fix does a check to confirm that if the file is found, it is not from the current path unless a relative or absolute path is provided.  The assumption being that having a relative or absolute path to the current directo [... cut]
closed 2021-08-06 by agramfort